speeduino/speeduino #290: Secondary serial issue (Mega2560 Serial 3)

muuris, 2019-12-07:
I'm developing my own incar controller, which need to have two-way communication with Speeduino. 

I've tried to opimize my controllers timing to have the least stress on Speeduino, which tries to read secondary serial rx every 64 loops and polls aux channels at 4Hz. I read the aux polls and queue them, so I can reply with specific intervals, not too quickly one after another. I poll statusdata with 'n' command with 150ms interval at minimum, I can then read loopsPerSec from statusdata and hence try to optimize timings to get my data comms in the middle of the 64 loop period.

The problem is, if I have both status polls and Speeduino AUX polls set up, Speeduino is crashing after some time, secl bounces to 0 and TS communication is lost. Secl goes to zero before byte limit also when TS is not connected (reading stats via serial 3). I thought this was USB or signal issue, but now I noticed that many times (but not every time) just before the Speeduino crashes freeRam shows > 45000 bytes both in TS and serial 3. I've examined the Speeduino code and haven't yet figured why this problem is occuring.

This is now tested with 201911 Speeduino firmware and two separate Mega2560s (one Speeeuino and one of my own).

Autohome2, 2019-12-07:
It sounds like your external code is not waiting for complete messages from speeduino before requesting more. This is a common issue that has been discussed before.
I'm not sure understand correctly but you are not only requesting data from speeduino but having it read external data too?
Without seeing code for how that is being handled it is very difficult to follow that error.
May I suggest looking at my gpio std firmware to give you some ideas on how to handle this ?

muuris, 2019-12-07:
Yes, my controller needs Speeduino realtime data (sends 'n' polls) and also acts as external IO (receives 'R' aux channel requests, queues them and replies with specified intervals, one channel request at a time). This problem doesn't come up if only polling realtime data or sending aux channel data.

I've checked the std gpio code, doesn't seem to work as is with 201911 firmware. I got mini gpio working and I think I managed to get the same problem. The gpio codes don't queue Speeduino aux polls but sends them instantly as the request is processed. I found this problematic in 2-way comms if using more than one aux channel, it worked better to queue requests and reply so that Speeduino receives only one channel request between the 64 loop period it checks secondary serial.
 
My controller polls realtime data from Speeduino, not more frequently than 150 ms. This is minimum time, longer if not enough time passed since sending Speeduino aux values. No polls are sent if there is aux data queue, no polls are sent within 50ms of the last aux reply, and aux replies are sent after >32 Speeduino loops has elapsed from last serial event.

I'll have another go at gpio but still, I think Speeduino should not crash from any secondary serial events, it should rather trash them. One problem I see is Speeduino getting aux reply ('G') but this is handled before the rest 9 bytes are present, then they are ignored as aux data but remain in serial buffer before next 64 loop cycle. Then one byte is read, and as it's not 'G', 'A' etc it's ignored, but only that one byte. So it takes 64*9 loops to get rid of the trash, and before this there is new data coming..

If my controller gets 'n' or 'R', it expects the rest of data with a timeframe. It not read out properly, it reads serial buffer into bit space until it's empty.

pazi88, 2020-04-18:
@Autohome2 I have seen this issue also. It's actually quite easy to totally crash speeduino by using Serial3. Just by using "A" or "r" to read data from speeduino. Probably caused by Rx buffer overflowing. Is there way to prevent the crash from happening?

muuris, 2020-04-18:
Just as a quickly thought example, maybe clear the whole buffer if cmdPending -flag stays up for n cycles without actual processing of the buffer data?

noisymime, 2024-02-05:
Closing this as stale. Significant changes have been made to the secondary serial code since this time. If issues are still experienced, please open a new specific issue.

closed 2024-02-05 by noisymime