speeduino/speeduino #296: MAP reading spikes up while cranking

muuris, 2019-12-20:
At least since 201909 release the MAP reading has spiked up while cranking. This is nasty with big injectors as pulsewidth goes very high, unless VE table has been manipulated to avoid this (but that's really just a workaround, even though the VE table area in which this leads to is practically unreachable)

[image]
250kPa MAP in this, I've seen this effect on all three of my Speeduino builds. MAP signal is fine after initial peak. Haven't had time to investigate this, as the workaround gets away with it.

gilberninvader, 2019-12-20:
I also get exactly the same. It spikes up as Sync is established. I have only tested on the bench so far so don't know what effect it may have when on the car. I reported this in November as part of issue #199. I still have the enduring problem that sync takes many attempts to become established when any level of trigger filtering is selected.

DeeEmm, 2019-12-20:
I can confirm the same behaviour: MAP spikes whilst cranking.

noisymime, 2019-12-21:
Just to confirm, which MAP sampling method do you have selected when seeing this?

iLeeeZi, 2019-12-21:
I tested it on bench and it happens with Cycle average and Event average. I don't remember it doing it few months ago when I was using Cycle average but not 100% sure.

DeeEmm, 2019-12-21:
> Just to confirm, which MAP sampling method do you have selected when seeing this?

I'm using Cycle Average.

muuris, 2019-12-21:
Cycle average in mine too.

noisymime, 2019-12-21:
Should be fixed in master now. 

Please try either compiling from Github or using the `master` firmware selection in SpeedyLoader if you want to test.

closed 2019-12-21 by noisymime

lsdlsd88, 2020-05-21:
this is still an issue on 202002. attached a log and tune used for it. occasionally it reads bad rpm and bad map. onboard sensor on mx5 pnp so I can't see how it would be possible to read 400kpa! clearly a software bug. 
problem occoured at 22.3 seconds
[map spike.zip](https://github.com/noisymime/speeduino/files/4661512/map.spike.zip)

noisymime, 2020-05-21:
@lsdlsd88 The MAP spikes in that log are related the RPM spikes, my guess is that you're getting some electrical spiking or noise that's causing you to see this as the two would otherwise not be occurring together like they are at 22.3. It's extremely unlikely to be related to this issue. 

Try setting your MAP sampling to Instantaneous and see if you get the same issue.

lsdlsd88, 2020-05-21:
I thought about that but how's possible to read a value higher than the maximum adc reading ?

noisymime, 2020-05-21:
Because you're using the Event Average, the end MAP value you get is using the ignition event count as one of it's variables. If the RPM has gone crazy, the ignition event count will also spike and affect the MAP reading.

The cause of the issue still seems likely to be a problem with the RPM, but it does raise a valid point that there should be a safety check there for the MAP reading

lsdlsd88, 2020-05-21:
understood. will try with instantaneous and also take some composite log to see whats causing the rpm spike

commit 1c4a6b6f (2019-12-21): Prevent MAP spike when cranking begins with averaged sampling. Fixes #296