speeduino/speeduino #299: Portablity to other CPUs ?

fluke9, 2020-01-01:
I would like to port the complete speeduino code to C167 and therefore almost all ME7 Bosch ECUs.
Although i have completley reverse engineered a ME7 and labeled almost anything and also understand most of the torque structure, DBW control and so on and can binary patch almost everything i want into it i would really like to have something which can be compiled and changed.

So maybe you can answer me a few quick questions which will help me finding out if porting is possible at all or if should just start from scratch:
- How much does speeduino rely on the HW ? Timers ok, no problem, you got plenty of them on the C167.
- Is that STM32 port in a working condition ?
- 512kb FLASH and 1kb of EEPROM are available (atleast, some have more), how much space does a common config take, we could take a flashpage or just upgrade the EEPROM...
- Crank sync works with simple interrupt code ?

Other things is have to work out is if the communication protocol can be made K-Line compatible.
What i can offer in turn to do is:
- Writing code to control DBW throttle bodys, a TLE7209 is used which is a H-bridge with simple status feedback.
- Writing code for MAF support, there is lot of trickery in there which i now mostly understand.
- Writing the drivers for other stuff.

Besides from the connectors, a VW or AUDI ME7.x.x is probably the cheapest sophisticated ecu hardware you can get for a cheap buck used.

VitorBoss, 2020-01-02:
The STM32 port is working on bench, no engine was fired by it yet, it isn't full stable at moment, after recent core update it have some work to do and I'm on it.

noisymime, 2020-01-03:
Sounds like the EEPROM is going to be the limitation on the hardware side. The tune and calibration tables combined need at least 8KB, but for future proofing you'd ideally want more. That said, ensuring full compatibility with the Winbond W25 line of SPI flash is fairly high on my priority list at the moment, so I'd strongly suggest looking at sticking on of them on for expansion. 

Otherwise, the only real hardware requirements are having sufficient 16-bit timers (and compare channels) plus a sufficient number of inputs that are enabled for interrupts (I'm not familiar with the C167, all the pins might be enabled for this?)

Currently it's the Kinetis K64/K66 line of MCUs that is best supported (Outside of the AVR), but getting solid support for the stm32f407 line will continue happening as well.

fluke9, 2020-01-07:
> Sounds like the EEPROM is going to be the limitation on the hardware side. The tune and calibration tables combined need at least 8KB, but for future proofing you'd ideally want more. That said, [...]
I thought either of expanding the EEPROM by replacing it, which should be not a big deal, its just a tiny 8 pin SOIC.
Or dedicate a flash sector with a a primitive wear leveling algorithm to store the data,
so it could run on unmodified stock hardware. Most ECUs have either a 29F400 or 29F800 which means 512kb or 1024kb. 

> Otherwise, the only real hardware requirements are having sufficient 16-bit timers (and compare channels) plus a sufficient number of inputs that are enabled for interrupts (I'm not familiar with the C167, all the pins might be enabled for this?)
The C167 is quite capable in this regard, almost all pins can be used as interrupt (besides the adcs) or as trigger for the the CAPCOM (capture compare) units which can do a lot of legwork and provide interrupts themselfes. Lots of timers available too. The OEM hardware will have the HW wired to the correct pins for sure ;-)

> Currently it's the Kinetis K64/K66 line of MCUs that is best supported (Outside of the AVR), but getting solid support for the stm32f407 line will continue happening as well.
The reason i aske was mainly because if there are ports its more likely to be portable code ;-)

noisymime, 2022-01-05:
Closing as stale. Please open a new issue if still required

closed 2022-01-05 by noisymime