speeduino/speeduino #300: Improvements to the GM 7X decoder

noisymime, 2020-01-03:
This is a tracking item to improve the current state of the 7X decoder.

twisted-builds-llc, 2020-01-18:
Verified with timing light that Cylinder 1 is 5* retarded ignition timing vs commanded. Found the Trigger Angle (Deg) setting under Trigger Settings in tuner studio does not allow this to be corrected at this point in time for machining variance between crankshaft wheels. Also verified OEM 4200 computer has a "crankshaft relearn" procedure so the factory PCM can learn the machining variance between factory crankshafts, so having the ability to dial in the trigger angle would be nice.

Also believe timing is off between cylinder firing orders at this time. However no definitive proof can be given at this time.

Attached is the 4200 tune used during a dyno pull. This seemed to be 100hp down from when a known good ECU base was used. Same target and achieved AFR and Boost was verified between pulls between ECU swaps. Will update with more information when it becomes available. 
[Calvin Nelsons 4200 Speeduino Tune.zip](https://github.com/noisymime/speeduino/files/4081208/Calvin.Nelsons.4200.Speeduino.Tune.zip)

noisymime, 2020-02-06:
Reference image: 
[image]

This diagram is what was used to determine the 42* fixed angle on the original decoder design

noisymime, 2020-02-06:
I've added the ability to change the trigger angle now on this pattern. It must currently be below 48 and above -81, but hopefully that should be enough range for your testing if it's only 5* out.

One thing to check would be which edge you've got set. If you're triggering on the wrong edge (or I've coded for the wrong edge), this could cause a difference of around 5* as that's approximately the width of the tooth. 

Everything in my testing seems to work ok, but it is possible that the stim I'm testing against isn't quite right. Please make sure you've got the new ignition mode (In Spark settings) turned on.

knockoutperformance, 2020-03-02:
I have an gm 3900 v6 from an 09 Impala that I'm planning to use in a resto-mod. This unit had a 58x (60 teeth missing two) and a weird 4 notch in the cam for triggers. Could it be possible to read a slightly different 12 degree sync notch on an 7x trigger wheel? I can grind off the teeth on the 58x wheel to have an 7x but with an 12 degree sync pulse slot instead of 10 degrees. (58x @ 6000rpm, I feel would be a bit too much for the mega2560). The camshaft I can put an old dizzy in and set that up for the single signal. Thank you for your time.

pazi88, 2020-03-02:
@knockoutperformance Maybe the cam trigger wheel is this one: https://github.com/noisymime/speeduino/issues/327

noisymime, 2020-03-02:
@knockoutperformance I would 100% be using the 58X (60-2) pattern you've got. It's very well supported and gives significantly better resolution than the 7X. 

Many folk running 60-2 on Speeduino (commonly BMWs) without issues. The only change potentially required is to the filter capacitor on the crank input.

knockoutperformance, 2020-03-02:
@pazi88 yes that is the camshaft style I have in this engine. It does have vvt , but I am not going use, have set that to 0 degrees and locked it. Thank you

@noisymime ok thank you! I'm going to have to find another 58x, the one I have has been damaged, looks like at some point a screwdriver was used to roll engine over using the 58x, prolly why it was in junk yard.