speeduino/speeduino #302: Possibility of Rotational Idle?

tommmtom, 2020-01-12:
Would it be possible to add in rotational idle? 

Activated within set CLT and TPS values

Would be a great addition!

requies, 2020-08-08:
I second that request. Would be really nice.

requies, 2020-08-09:
I second that.. I'd really like this feature.
Activated within set CLT and TPS values or just an input pin.

DeeEmm, 2020-08-09:
I agree. Got my vote too 👍

Thommmm, 2021-01-05:
To clarify, what is rotational idle?

requies, 2021-01-05:
Rotational idle / roving idle its where the computer cuts ignition to one of the cylinder in a rotational  sequence (usually round-robin), in order to allow the engine to idle with large throttle openings and to keep temperatures down after using antilag.

quote from perth-wrx.com (BALISTC)

> Anti-lag systems retard the ignition timing and, in combination with a throttle opening system such as a solenoid, idle speed motor or throttle screw, open the throttle wide open, so unburned fuel and a large amount of air pass through the engine and, instead of igniting in the engine, the fuel/air ignites in the exhaust manifold, and spins the turbo over, even at low engine speed.
The theory is, as soon as you open the throttle, you have full boost and no lag.
The air needed for this combustion is provided by the standard throttle body, which is opened to a greater extent than normal whilst anti-lag is running...
The way the throttle is held open is a discussion within itself. On Autronic, they use the idle speed motor, where as on other ECUs such as PossumLink or any ECUs which dont have enough outputs to run an idle speed motor, a throttle screw or solenoid is used to keep the throttle open (as stated by Group N rally rules). With a Motec, either the idle speed motor or a solenoid can be used. Running a throttle screw or solenoid allows more air to flow in, hence a greater anti lag effect.
On my car, I have a VR Commodore air-con solenoid to hold the throttle open. Having a constant throttle opening with a throttle screw is not convenient on a street car.....throttle screws are fine on rally cars, as driveability isnt their main concern.
The ECU uses an idle control sequence so that once you've set the target idle speed, the anti-lag will cut a spark in a rotational sequence with each cylinder, which is why they idle so lumpy, and thus the reason its called rotational idle. If there was no rotational idle, the engine wouldnt idle at all, because the throttle is so far open that the engine would just rev.
The rotational idle's main task is to keep temperature under control after running anti-lag. The anti-lag system increases EGTs, and in race applications it can compromise the exhaust manifold, turbo and exhaust valves. The rotational idle cools everything right down after a run, and if you feel the engine bay, you can actually feel how much its cooled down if you leave the rotational idle on for a while.
The anti-lag system activates on most ECUs as soon as the engine RPM goes over the arming target. When this occurs, the anti-lag goes into full anti-lag mode, and it pops and bangs very loudly. The size of the bangs and pops depends on the amount of ignition retard. Once the revs come back down, it idles very loudly at about 2000rpm. As soon as the engine is placed under any load, even just cruising, there is positive boost pressure.
If the RPM doesnt meet the RPM arming target for a certain amount of time (user definable on some ECUs, preset on others IE 15 seconds), the anti-lag switches back into the rotational idle until the revs are increased to the arming target once again.
The anti lag is NOT a good street modification. When the engine is in anti-lag stand by mode, the idle is lumpy and makes the whole car shake, not to mention it being super loud, although it does sound awesome.
The actual engine is MUCH louder with the anti-lag turned on, because of the massive air flow, its pops and jumps around at low speed, and if you go over the anti-lag rpm threshold, the car gets even LOUDER, and pops flames. It can make the brakes very hard due to the lack of vacuum, and it can catch you out if your not careful.
To take off, you have to bring the revs up or it'll stall and buck around, which isnt good for the clutch or gearbox. As already stated, lots of throttle is needed to keep the car driving smoothly.
Different story at the track though. It produces tyre smoking launches, boost comes on instantly, and there is no need for flat changing or anything like that, because the anti-lag does all the work for you keeping the engine on boost.
As for turbo longevity, rally cars run incredibly aggressive anti-lag 100% of the time, but for many teams, replacing engines, exhaust manifolds and turbos isnt an issue.
On a street car, with occasional use, EGTs arent a huge problem, unless you go onto the freeway, rev the engine out and back off the throttle...high RPM over run for constant periods of time with the anti-lag switched on is what damages turbos and exhaust manifolds, because of excessive EGTs.
Some high end ECUs have an EGT recovery strategy, that reduces or disables anti-lag when EGTs exceeds the set figure, which means you save your exhaust manifold, exhaust valves and turbo from melting into a heap.