speeduino/speeduino #305: Cranking enrichment range

muuris, 2020-01-16:
I've found the cranking enrichment range is not enough for colder climate country and pure E85 use. I think instead of a byte modifier pct it should be either a byte adder pct (to 100% base) or an int modifier value. From what I've read even some stock ecu maps for gasoline use ~300% modifier for freezing temperatures (0°C), which is about what I need in mine for E85.

Modifying it to a byte adder would allow 100% more enrichment but would require modifying every tune and every user should understand the new principle, so the int modifier would be better option. The table would need to moved to the end of config10, but there's room.

pazi88, 2020-01-16:
There is also 255% limitation on gamma enrichment which can become problem if when only cranking enrichment is modified. But I agree. Wider enrichment range is needed for cold environments.

muuris, 2020-01-16:
True, gamma enrichment is a limiter as well.

muuris, 2020-01-18:
Another option (which I'm now running on mine) is to use the old byte value, but use it with 5 % increments, which would allow maximum of 1275 % cranking enrichment. I think 5 % precision is enough for cranking enrichment, and as it is kinda exception to other enrichments with it's huge max value, it could be checked outside correctionsFuel() if gammae/corrections value is wanted to retain as a byte. The downside would be of course that actual enrichment can't be seen during cranking. This is just a way to minimize modifications necessary. After update the old values just need to be loaded from the TS project to Speeduino.

I'm willing to be assigned for this, if Josh approves and points out how this is wanted to be done.

noisymime, 2020-01-22:
Just a little update. I've made the changes for GammaE to be extended to 511% for now and it can go out beyond that without too much difficulty. I'll look at adding the changes specifically for cranking shortly.

muuris, 2020-01-22:
That's a good start!

Now that I've overcome the E85 long cold cranking issues (lack of fueling) I'm now using this cranking enrichment curve (byte, 5% precision)
[image]

..and getting 510% GammaE at 0°C cranking, so more is needed for cold environments. It will eventually start with less, but cranking will be significantly longer than normal.

The way I modified mine is using less bitshift in PW-function when corrections are >500% to avoid overflow. Loses a bit of precision, but only when the enrichment is huge anyways (should only affect very cold cranking basically).

closed 2020-03-16 by noisymime