speeduino/speeduino #311: Building Speeduino for STM32 target fails

squarecode, 2020-01-25:
First of all: Thanks for all the work that has already been done to this project!

I'm trying to compile the current release of speeduino for stm32 targets (bluepill_f103c8 and generic). After downloading and loading the sources in vscode I tried to compile them by clicking Terminal ->Run task ->Platformio build stm32...
However it seems that I'm missig a library. A quick google search turned mbed targets as the source of the missing file up. So I was having a look at the platformio.ini to look for the dependencies. Framework is set to Arduino, so this was the first thing that threw me off. I was expecting mbed as the missing file was part of it. Also the lib_deps are nothing special. The dependency search from platform io as mentioned in the compiler log failed to find anything matching. 
```
speeduino/board_stm32_official.h:7:10: fatal error: stm32f4xx_ll_tim.h: No such file or directory

**************************************************************************
* Looking for stm32f4xx_ll_tim.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stm32f4xx_ll_tim.h"
* Web  > https://platformio.org/lib/search?query=header:stm32f4xx_ll_tim.h
*
**************************************************************************
```
Clearly I must be doing something wrong at this point. This is my second contact with platformio so far so stupid pebkacs are to be expected, however I have a lot of experience with the stm32 lineup.

Hopefully someone can push me in the right direction.

VitorBoss, 2020-01-25:
You have the latest version of the [Arduino_Core_STM32](github.com/stm32duino/Arduino_Core_STM32#)?

squarecode, 2020-01-27:
Thanks for your fast respone!
I might be misunderstanding you, but you mean the usual st stm32 platform and arduino framework? For my understanding I don't have to clone manually git repos for the librarys and it will break platformio because the json files for it to work with are missing. Btw your link is broken...

I tried it now with uninstalling the st stm32 platform and reinstalling (version 6.0.0). After that I reloaded the environment and tried compiling again. But the file is still missing. I could of cause clone the stuff and get it to work, however that is not the way it is supposed to work i guess so I'm happy to do it the right way first.
I've included the entire build output below, however the interesting part is likely this one:

```
Processing bluepill_f103c8 (platform: ststm32; framework: arduino; board: bluepill_f103c8)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing framework-arduinoststm32 @ ~3.10700.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
framework-arduinoststm32 @ 3.10700.191028 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 6.0.0 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32 3.10700.191028 (1.7.0)
```

<details> 
  <summary>Entire output</summary>

```
Processing bluepill_f103c8 (platform: ststm32; framework: arduino; board: bluepill_f103c8)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing framework-arduinoststm32 @ ~3.10700.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
framework-arduinoststm32 @ 3.10700.191028 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 6.0.0 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32 3.10700.191028 (1.7.0)
Converting speeduino.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 2.0.1
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/bluepill_f103c8/FrameworkArduinoVariant/PeripheralPins.c.o
cc1: warning: command line option '-fpermissive' is valid for C++/ObjC++ but not for C
Compiling .pio/build/bluepill_f103c8/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/Print.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/RingBuffer.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/USBSerial.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/WInterrupts.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/WSerial.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/WString.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/abi.cpp.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/board.c.o
cc1: warning: command line option '-fpermissive' is valid for C++/ObjC++ but not for C
cc1: warning: command line option '-fpermissive' is valid for C++/ObjC++ but not for C
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/hooks.c.o
cc1: warning: command line option '-fpermissive' is valid for C++/ObjC++ but not for C
Compiling .pio/build/bluepill_f103c8/FrameworkArduino/itoa.c.o
[... 361 lines cut ...]
**************************************************************************

 #include "stm32f4xx_ll_tim.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/bluepill_f103c8/src/speeduino.ino.cpp.o] Error 1
==================================================================== [FAILED] Took 45.94 seconds ====================================================================

Environment                      Status    Duration
-------------------------------  --------  ------------
megaatmega2560                   IGNORED
teensy35                         IGNORED
teensy40                         IGNORED
genericSTM32F103RB               IGNORED
genericSTM32F103RB_STM32GENERIC  IGNORED
black_F407VE                     IGNORED
bluepill_f103c8                  FAILED    00:00:45.939
samd21                           IGNORED
=============================================================== 1 failed, 0 succeeded in 00:00:45.939 ===============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```
</details>

VitorBoss, 2020-01-27:
I think the framework is just ststm32
The core used is:
github.com/stm32duino/Arduino_Core_STM32

squarecode, 2020-01-27:
For the platform-ststm32 I got:
```
➜  speeduino-201911 pio platform install "ststm32"
PlatformManager: Installing ststm32
ststm32 @ 6.0.0 is already installed
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0
toolchain-gccarmnoneeabi @ 1.70201.0 is already installed
The platform 'ststm32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
```
The settings in the platformio.ini are

```
[env:bluepill_f103c8]
platform = ststm32
framework = arduino
; framework-arduinoststm32
board = bluepill_f103c8
lib_deps = EEPROM
;build_flags = -fpermissive -std=gnu++11 -Os -ffunction-sections -fdata-sections -Wl,--gc-sections -Wl,-Map,output.map
build_flags = -fpermissive -std=gnu++11 -Os -DCORE_STM32_OFFICIAL -UBOARD_NR_GPIO_PINS
```

And with framework-arduinoststm32 I get:

```
Processing bluepill_f103c8 (platform: ststm32; framework: framework-arduinoststm32; board: bluepill_f103c8)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 6.0.0 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Error: This board doesn't support framework-arduinoststm32 framework!
```

I'm really confused at this point. I tried also setting the links from your post into the ini file, however I got the missing lib error again. So same behavior as with the already installed ones...

VitorBoss, 2020-01-27:
> Processing bluepill_f103c8 (platform: ststm32; framework: framework-arduinoststm32; board: bluepill_f103c8)

You are still using arduinoststm32 there

squarecode, 2020-01-28:
It's the default config:

```
[env:bluepill_f103c8]
platform = ststm32
framework = arduino
; framework-arduinoststm32
board = bluepill_f103c8
lib_deps = EEPROM
;build_flags = -fpermissive -std=gnu++11 -Os -ffunction-sections -fdata-sections -Wl,--gc-sections -Wl,-Map,output.map
build_flags = -fpermissive -std=gnu++11 -Os -DCORE_STM32_OFFICIAL -UBOARD_NR_GPIO_PINS
```

What do I have to change in order to get it to work? I can't enter what you said above, because the framework=/=platform.

> I think the framework is platform-ststm32
>The core used is:
>github.com/stm32duino/Arduino_Core_STM32
>Edit: https://github.com/platformio/platform-ststm32

What I tried so far was:

1.  to set the framework to framework-arduinoststm32, because it was commented out (of cause only one of them was enabled at a time). 

2. For the platfrom I tried platform-ststm32. And also the combinations of the two. 

3. I also tried board_build.core = stm32 and board_build.core = Arduino_Core_STM32

Thats why I don't understand what you mean. I tried all combinations that made sense to try from my standpoint.

VitorBoss, 2020-02-01:
Now I realize what is it. That file was removed some time ago. I've submitted a PR removing that line and fixing some other parts of the code. You can try from my fork and see if is working?

@Tjeerdie Did you compile the code using PIO?

VitorBoss, 2020-02-01:
Keep in mind that framework is behind the official core and some boards and functions may not work.
See this PR https://github.com/platformio/platform-ststm32/issues/345

Tjeerdie, 2020-02-01:
> Now I realize what is it. That file was removed some time ago. I've submitted a PR removing that line and fixing some other parts of the code. You can try from my fork and see if is working?
> edit: https://github.com/VitorBoss/speeduino/tree/noisymime-master
> 
> @Tjeerdie Did you compile the code using PIO?
> 
> EDIT 2: Try this:
> 
> ```
> platform = ststm32
> framework = arduino
> board = bluepill_f103c8
> lib_deps = EEPROM
> build_flags = -fpermissive -std=gnu++11 -Os -ffunction-sections -fdata-sections -Wl,--gc-sections -Wl,-Map,output.map
> platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32
> ```

I tested my code using both PIO and arduino before making a PR. The last time a made a PR the official core was not up to date with the new Timers yet. And therefore to compile for the STM32F4 a specific version of the official core was selected in the plaformio.ini for the STM32F4 board in my PR I did not have a look at the code recently. When i get my iteration 2 board for testing i will try and fix any oustanding issue regarding compiling

squarecode, 2020-02-04:
Thank you so much for your time! I've tried your PIO settings as well as the speeduino fork from you Vitor (with and without the edited build settings. It fails again but this time with a missing lib for EEPROM. I'll have to take a look at this (probably this evening) and report back as this is probably a setup issue on my side.
<details>

```
Looking for EEPROM library in registry
Conflict: More than one library has been found by request {"name": "EEPROM", "requirements": null}:
eeprom
======
#ID: 2284
I2C EEPROM library from 24C01 to 24C1025

Keywords: eeprom
Compatible frameworks: Mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, RISC-V GAP, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Automatically chose the first available library (use `--interactive` option to make a choice)
Authors: bernard borredon

EEPROM
======
#ID: 3516
I2C EEPROM library for 1k to 512k chip

Keywords: c24lcxxx, eeprom, i2c
Compatible frameworks: mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Authors: Wojciech Biegun

eeprom
======
#ID: 2287
STM32 EEPROM emulation AN2594

Keywords: eeprom, flash, stm32f10x
Compatible frameworks: mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Authors: Chau Vo

eeprom
======
#ID: 2286
LPC11U24 internal EEPROM

Keywords: eeprom, lpc11u24
Compatible frameworks: mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Authors: Suga koubou

EEPROM
======
#ID: 2201
EEPROM in TG-LPC11U35-501

Keywords: eeprom
Compatible frameworks: mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Authors: yasuyuki onodera

eeprom
======
#ID: 3514
stm32l053 eeprom

Keywords: eeprom, read, stm32
Compatible frameworks: mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Authors: lzbp li

eeprom
======
#ID: 6378
Stm32L4 eeprom emulation library for reading and writing to flash memory to extend 10K erase/write cycle of wear-off

Keywords: stm32, eeprom, emulation, stm32l4
Compatible frameworks: STM32Cube
Compatible platforms: ST STM32
Authors: Abdulkadir Yaman, Cagatay Yucelen

eeprom
======
#ID: 2179
sf

Keywords: eeprom
Compatible frameworks: mbed
Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
Authors: li lei

Found: https://platformio.org/lib/show/2284/eeprom
LibraryManager: Installing id=2284
eeprom @ 925096a4c7 has been successfully installed!
Found 0 compatible libraries
```

</details>

VitorBoss, 2020-02-04:
The EEPROM needed is at https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/EEPROM

It's part of the core, the main is up to date with my changes.

Tjeerdie, 2020-02-04:
> The EEPROM needed is at https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/EEPROM
> 
> It's part of the core, the main is up to date with my changes.

Does speeduino work with this EEPROM emulator? Last time i tried the speeduino code did hang during burns because of slow writes. Is that fixed?

squarecode, 2020-02-12:
After a few days of fiddling around I am back. I uninstalled vscode and installed the newest version, I reinstalled plattformio and also deleted all previous fetched libs, frameworks etc. from my hdd. After reinstalling everything I tried a clean clone from Vitors fork and I still have the same issues... With the default bluepill config the build fails with the stm32f4 timer header missing and with the modified build config from Vitors suggestion:

```
platform = ststm32
framework = arduino
board = bluepill_f103c8
lib_deps = EEPROM
build_flags = -fpermissive -std=gnu++11 -Os -ffunction-sections -fdata-sections -Wl,--gc-sections -Wl,-Map,output.map
platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32

```

I fail with the EEPROM lib missing. The thing is there is no EEPROM lib that seems to fit the config from above. All the ones that platformio lists during compiling are for different frameworks. How can I ensure that I'm using the same libs and settings as you are using? I've tried various other things not worth mentioning in the last couple days as I really wanted it to work...

VitorBoss, 2020-02-15:
Sorry the delay, I have finally figured out the file 'stm32f4xx_ll_tim.h' is not available for BluePill, I have changed to include the 'stm32f1xx_ll_tim.h' for STM32F1 boards.

squarecode, 2020-02-17:
Thank you so much! It finally compiles, however the flash overflows now.
```

region `FLASH' overflowed by 34596 bytes

```

By quite a lot in my opinion...

noisymime, 2020-02-18:
> It finally compiles, however the flash overflows now.

The f103c8 boards (ie Bluepills) are inconsistent in the amount of flash they have. The majority appear to have 128kb (which is sufficient) but officially they only have 64kb (which isn't). Whilst they appear to be rare, there are bluepill boards out there with only 64gb flash and given it's the official figure, that's what the sketch is compared against.

pazi88, 2020-02-18:
> The f103c8 boards (ie Bluepills) are inconsistent in the amount of flash they have. The majority appear to have 128kb (which is sufficient) but officially they only have 64kb (which isn't). Whilst [...]

Yeah. Even when some bluepills should have 64k flash based on the chip number, those still usually have 128k of flash. And it can be used, just by choosing the variant which has 128k of flash before building.

closed 2020-02-18 by noisymime

squarecode, 2020-02-18:
Yeh... should have checked that first. I we are comparing against the 128k version... Thank you all so much for your help! I hope to be able to support this project in the future as soon as my f4 hardware is done.

noisymime, 2020-02-18:
I've changed the Bluepill config to force it to use the 128k variant. This now compiles cleanly, but will obviously have problems if you try to upload it to a 64kb board.

ArtCodeZen, 2024-07-30:
Hello I'm trying to compile the speeduino with (Bluepill 128k)stm32f103c8t6, with vscode platformIO and Arduino IDE, but there are some problem with timers:

static inline void FUEL5_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC1; (TIM5)->DIER |= TIM_DIER_CC1IE;}
static inline void FUEL6_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC2; (TIM5)->DIER |= TIM_DIER_CC2IE;}
static inline void FUEL7_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC3; (TIM5)->DIER |= TIM_DIER_CC3IE;}
static inline void FUEL8_TIMER_ENABLE(void) {(TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->CR1 |= TIM_CR1_CEN; (TIM5)->SR = ~TIM_FLAG_CC4; (TIM5)->DIER |= TIM_DIER_CC4IE;}

static inline void FUEL5_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC1IE;}
static inline void FUEL6_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC2IE;}
static inline void FUEL7_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC3IE;}
static inline void FUEL8_TIMER_DISABLE(void) {(TIM5)->DIER &= ~TIM_DIER_CC4IE;}

  static inline void IGN5_TIMER_ENABLE(void)  {(TIM4)->CR1 |= TIM_CR1_CEN; (TIM4)->SR = ~TIM_FLAG_CC1; (TIM4)->DIER |= TIM_DIER_CC1IE;}
  static inline void IGN6_TIMER_ENABLE(void)  {(TIM4)->CR1 |= TIM_CR1_CEN; (TIM4)->SR = ~TIM_FLAG_CC2; (TIM4)->DIER |= TIM_DIER_CC2IE;}
  static inline void IGN7_TIMER_ENABLE(void)  {(TIM4)->CR1 |= TIM_CR1_CEN; (TIM4)->SR = ~TIM_FLAG_CC3; (TIM4)->DIER |= TIM_DIER_CC3IE;}
  static inline void IGN8_TIMER_ENABLE(void)  {(TIM4)->CR1 |= TIM_CR1_CEN; (TIM4)->SR = ~TIM_FLAG_CC4; (TIM4)->DIER |= TIM_DIER_CC4IE;}

  static inline void IGN5_TIMER_DISABLE(void)  {(TIM4)->DIER &= ~TIM_DIER_CC1IE;}
  static inline void IGN6_TIMER_DISABLE(void)  {(TIM4)->DIER &= ~TIM_DIER_CC2IE;}
  static inline void IGN7_TIMER_DISABLE(void)  {(TIM4)->DIER &= ~TIM_DIER_CC3IE;}
  static inline void IGN8_TIMER_DISABLE(void)  {(TIM4)->DIER &= ~TIM_DIER_CC4IE;}

VitorBoss, 2024-07-30:
BluePill isn't supported anymore, the support was dropped in 2020 due to flash size limitations.

ArtCodeZen, 2024-07-30:
> BluePill isn't supported anymore, the support was dropped in 2020 due to flash size limitations.

Ohh, thank you!

commit 433de94f (2020-02-18): Try to fix #311 (#324)

* Some Fixes to work

* Fix BluePill

* Remove libmaple defs

* STM32 Timers rework

* GCC9 Fix

* BlackPill update

* Disable FRAM as default for STM32

* STM32 storage options for all cores

* STM32 fixes 2

* Update FRAM lib

* Fix previos ADC 'fixed definitions'

* Fix duplicated pins

* Typo

* Fix compilation error for BluePill

* Fix timers Prescale