speeduino/speeduino #322: IAC stepper motor ignored if temperature > max graph value

flagoworld, 2020-02-09:
If the current coolant temp is higher than the last dot on the graph, the IAC stepper position stays at what it was set to in the cranking table, instead of defaulting to the highest value in the graph.

ric355, 2020-02-10:
I presume you are referring to open loop operation.  I think what you have observed is deliberate. There is a specific check that verifies the current coolant temperature is less than the last coolant value in the running table;

`        else if( (currentStatus.coolant + CALIBRATION_TEMPERATURE_OFFSET) < table2D_getAxisValue(&iacStepTable, (IDLE_TABLE_SIZE-1)) )
`

Whether that is what is wanted for everyone or not is up for debate I guess, but it seems logical to me that the running table should cover all possible scenarios of coolant temp. So having the maximum coolant value less than the one in cranking is perhaps not the normal case.

flagoworld, 2020-02-10:
Other tables, such as warmup enrichment, various cranking tables, etc, seem use the last value on the graph whenever they are beyond that value. It seems inconsistent to do otherwise, especially without documentation about it. Is there a reason for doing it the way it is? A real-world use case? If not, it may be a good idea to change it so it does the same as other tables. If there is a real-world case, let me know and close the issue, no worries.

noisymime, 2020-02-13:
I agree here @flagoworld the logic to including this extra check probably isn't needed. This was originally in place as an efficiency check, but the actual table lookup has incorporated this now so it can be removed to prevent the exact case you're talking about

commit 1f7e75b6 (2020-02-13): Remove check on the OL Stepper idle for max temp. Fixes #322
closed 2020-02-13 by noisymime