speeduino/speeduino #327: Request: Bosch "Quick-Start" / VVT decoder

iLeeeZi, 2020-02-22:
[image]
 (Angles in the image might not be accurate for every setup)

This trigger wheel pattern has been used in many vehicles for last twenty years. It runs at camshaft speed usually with 60-2 crank wheel and it is used for phase synchronization and camshaft angle location. I'm not sure what its official name is as Bosch and Volkswagen calls it "Quick-Start sender wheel" and GM calls it "4x Reluctor" and I have seen it used by Audi, BMW, Ford, Volvo Etc. Mainly in vehicles with variable valve timing and Bosch ECU.

I made a trigger pattern for ardustim as I don't currently have the trigger wheel installed in my car. The following waveforms should be pretty close to VW/Audi 1.8T and VR6 Engines without VVT.
[image]
[image]

I have seen few requests in forums for this pattern and as I'm going to be using it in my vehicle I did a bit of testing and it seems to be pretty easy to add to Speeduino for phase detection. 

As this is used mainly with 60-2 crank wheel the phase can be detected by checking if cam signal is high or low at certain crank wheel tooth. Most waveforms that I could find shows that it could be checked at the first tooth after missing tooth but a selectable tooth might be the best to support larger amount of setups.

Another option is to make a real decoder but it is of course a bit more of work but might be useful in the future.

pazi88, 2020-02-22:
This would be nice addition. As far as I know, this is only used in 4-cyl engines? But adding this there wouldn't be need to cut off the extra teeth anymore.

iLeeeZi, 2020-02-22:
VW used it in 6 cylinder vr6 engines from 1998 onwards and GM seems to use it with Gen4 V8 LS engines. Volvo used it in some 5 cylinder engines too. I spent almost a whole day trying to find some datasheets and it seems to be used almost everywhere. Yeah It would be good to add as some trigger wheels aren't as easy to cut to be used as 4-1 or single tooth trigger.

Titiz67, 2020-03-16:
> This would be nice addition. As far as I know, this is only used in 4-cyl engines? Or at least I haven't seen this in 6-cyl ones. But adding this there wouldn't be need to cut off the extra teeth anymore.

Hello, I found this for bmw, the most recent models of bmw e90, e60 and e81,
Uses this camshaft wheel, it looks a lot like that of vw's  . Ref :  11367578877
[image]

Titiz67, 2020-03-16:
And this for E46 M3 S54 Intake & Exhaust Camshaft Sensor Triggers for Aftermarket Management Systems .

[image]
 This is oem : 11317832199  
[image]

FL0WL0W, 2021-09-02:
gm v8 log of the pattern here along with 60-2 crank
[image]

5peedmanual, 2023-08-09:
How is this working out? I have this trigger pattern for my car.

mike501, 2023-08-11:
I do not believe anyone is actively developing this trigger pattern. If the pattern is used on a hall sensor I think it will work now if you select trigger poll. I don't think it will support VVC under trigger poll.