speeduino/speeduino #341: 4B11 Trigger Support

ElDominio, 2020-03-22:
[image]

Add support for the 4B11 crank trigger, seems to be 36-2-1 , camshafts are already half moon, so sequential support shouldn't be too hard.

ElDominio, 2020-03-22:
https://www.bountysource.com/…/90379627-4b11-trigger-support bounty posted!

metacomgd, 2020-04-23:
Working on it, btw the correct link is 
https://www.bountysource.com/issues/90379627-4b11-trigger-support-50

metacomgd, 2020-04-23:
Working on sequentiual, but before anything else I need to know which direction the wheel spins relative to the engine in the above picture.

ElDominio, 2020-04-23:
[image]

Here you go sir

mike501, 2020-11-18:
@metacomgd any update on progress? If not I'll look to take over in a few weeks and split the bounty.
@ElDominio are you still after this?

ElDominio, 2020-11-19:
There seems to be an error in the getRPM on this pattern, and there is no sync support. 

I wrote some code to fix a few things but i get no RPM.

[4b11modstoinitanddecoders.zip](https://github.com/noisymime/speeduino/files/5563838/4b11modstoinitanddecoders.zip)

I attached two updates I did to this code, but it's  still not working. I have the test car ready to go so please let me know if you can test

closed 2026-11-14 by ElDominio