speeduino/speeduino #369: 200202 release

TradieUte, 2020-04-27:
Setting up for 8 channel sequential. In function calculateIgnitionAngles(), calculations for 6 cylinder, channels 4 to 6 are missing; 8 cylinder, channels 5 to 8 are missing.

noisymime, 2020-04-27:
This looks like it's resolved in current master. Have you tried the master branch?

TradieUte, 2020-04-27:
Sorry, should have checked. Mind in isolation mode.Sent from Samsung tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/noisymime/speeduino/issues/369#issuecomment-619718651",
"url": "https://github.com/noisymime/speeduino/issues/369#issuecomment-619718651",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

noisymime, 2020-05-04:
Closing as this is resolved in master

closed 2020-05-04 by noisymime