speeduino/speeduino #383: Cooling Fan disable on Cranking

joelgross123, 2020-05-17:
I have speeduino on a 383 stroker small block Chevy.  I have multiple cooling fans to cool my sandcar in a rear-mount radiator configuration.  Currently when the cooling fans are enabled it draws alot of current.  When the engine is already at temperature and the engine is restarted, having the cooling fans enabled at key on (speeduino boot-up) complicates things.  I would like to see a feature to disable cooling fans during cranking mode until the engine has had a chance to hit idle rpm.  This would be a big advantage to my application.  Also this may help others tune in getting engine to stable idle rpm before large current load is applied.

noisymime, 2020-05-18:
[image]

Option added

closed 2020-05-18 by noisymime

joelgross123, 2020-05-18:
Thanks Josh. This looks perfect!

Joel

commit 05022a5d (2020-05-18): Add option to disable fan whilst cranking. Fixes #383