speeduino/speeduino #384: Fuel Pressure & Oil Pressure monitor

Kojacknet, 2020-05-18:
Hi Josh,

Thank you for your conference yesterday on Youtube. 
I'm the one who was asking for monitoring fuel pressure and oil pressure monitoring on Speeduino.
Therefore I post here as a request

NY-17, 2020-05-18:
Hi Josh,

Can we take this further and build it into a Generic sensor analog(0-5v) input feature? 
This could make it more configurable for everyone to input whatever channels they want to log like temperature, pressure, egt, etc.
We could have have 4 user input fields for Channel name(input alias), input unit, 0v value and 5v sensor output values.
The interpolation between 0-5v would be linear.

Example:
Input Alias - Fuel Pressure
Unit - Psi
0v value - 0 psi
5v value - 100 psi

pazi88, 2020-05-18:
All of this already is there in speeduino using local auxilary input channels. You can add up to 16 extra analog sensors and give those own aliases that are visible in TS and recorded in log. Only thing that the scaling to correct values need to be done in TS, which works ok. But then if you want to tie in some safety features to these like oil/fuel pressure safety, then the sensor scaling would need to be calculated in Speeduino rather than TS. So only thing left to do for this issue is to add sensor scaling that is calculated in Speeduino.

NY-17, 2020-05-18:
Addition of sensor scaling would be really useful, It could be used for safety features/error light or generic output features in future.

Kojacknet, 2020-05-19:
> Hi Josh,
>
> Can we take this further and build it into a Generic sensor analog(0-5v) input feature?
> This could make it more configurable for everyone to input whatever channels they want to log like [...]

I can't more agree to this if it's possible.

> All of this already is there in speeduino using local auxilary input channels. You can add up to 16 extra analog sensors and give those own aliases that are visible in TS and recorded in log. Only [...]

Hi Pazi, 

Would you please refer to me some tutorials or something that I can read to do it?
I have no knowledge on coding currently but I willing to learn.

pazi88, 2020-05-19:
@Kojacknet Here's tutorial how to setup extra analog inputs to Speeduino using current FW: https://speeduino.com/forum/viewtopic.php?f=14&t=2809