speeduino/speeduino #385: Blended MAP and TPS acceleration enrichment

Hillmanjames, 2020-05-19:
I would like to request that we can blend the MAP and TPS based acceleration enrichment in a similar way to how you can on a Megasquirt. I have attached screenshots below of what we currently have and how the mega squirt does it. It would help me greatly getting my ITB's smoothed out (either has a stumble coming off slight throttle using TPS based AE, or can't add enough in when required above 100kpa when in MAP based AE. A blended mode sorted this out on my last build)

Current Speeduino way of doing things:
[image]

How it works on megasquirt:
[image]

Thanks in advance, James

theelichtje, 2020-09-24:
Would also love this feature for the same reason! (ITB response)

cyrusthevirus33, 2020-10-05:
This can also be very useful in turbo applications, as when boost "hits", it pushes air in, acting like a throttle.

norwich2817, 2023-03-19:
I would also like this feature!