speeduino/speeduino #386: Idle RPM Delta and closed loop timing advance idle control

joelgross123, 2020-05-20:
With the recent addition of idle control via timing advance/retard, the offsets appear to work in reverse of what is documented in the wiki manual.  Per the manual, negative Idle RPM Delta should mean that the engine is idling below target idle.  Instead per log viewer traces the opposite is indicated.  When engine RPM is above target idle a negative Idle RPM Delta is indicated.

[image]

Thus to get the function to provide closed loop control the opposite control offsets have to be entered into the configuration screen.  See attached trace and config figures.

[image]

The solution would suggest the need to negate the Idle RPM Delta in the code and then the suggested WIKI documentation for the configuration slope would be correct.

pazi88, 2020-05-20:
I have noticed that too. Easiest solution would be just change the TS ini so that it will multiply the delta values by -1

joelgross123, 2020-05-20:
@pazi88 would that also then fix the sign on the log viewer reported Idle RPM Delta ?