speeduino/speeduino #397: Support for Ducati 48-2-2 Trigger Wheel

SaulGoodman1337, 2020-06-09:
Hi,
is it possible to build in this trigger?
Installed e.g. in a Ducati Hypermotard 796. 

[image]

If I have understood correctly, this Trigger Wheel does not work with Speeduino, is that correct? How much effort would it be to install it?

minceheid, 2020-06-21:
Your trigger wheel turns at cam speed, so you can configure it as a 24-2 crank wheel. Unless you have a method of determining which one is the compression stroke you'd need to run it as wasted spark. 

From your trace, it looks like the OEM ECU is figuring out the engine phases somehow though.

Phoenix721, 2021-02-17:
Configure it as 24-2 Crank Trigger Wheel, with Falling Sensor Edge, based on your Oscilloscope image. Swap your crank sensor wires around and then it will be Rising Sensor Edge, which I would prefer for this motorcycle engine.
Ignition Sequence is:
270 degrees - Cyl #2, then 720 degrees - Cyl #1
or
270 degrees - Cyl #2, then 0 degrees - Cyl #1

Remember to calibrate/sync your Crank Index Number with a timing light on Cyl #1 (due to Wasted Spark, this must be done with a fixed 0 degree ignition advance value).

Run Multipoint Injection...and Wasted Spark Ignition (Falling Edge).

With all of the above said. You may need to machine two extra teeth per side away. Counter-clockwise...two teeth away downward from the left and two teeth away upward from the right, when looking at the image you posted. You would have to swap this around, if you make the Trigger Sensor Edge to Rising.

mike501, 2021-08-15:
If the previous advice that your trigger wheel is at crank speed is incorrect then 48 teeth doesn't divide into 360 without a remainder (its 7.5). You'll need some custom code as speeduino can't handle any remainders. It shouldn't be difficult as it just needs to ignore every other tooth making the pattern appear to the system as 24-1-1. You then need to use the cam to tell the system which tooth you're on in the 720 degree cycle and its done. 

Im just finishing off a couple of patterns. Give me a nudge in a couple of weeks if the advice above didn't work and you do need custom code as the pattern is running at crank speed.

denandz, 2021-09-17:
Hiya. This issue is pretty far up the list when searching for 'Ducati trigger pattern', so I figured I'd add some info. This wheel spins at cam speed and is effectively a 24-2 missing-tooth crank-speed trigger. The wheel seems to be used in most modern air-cooled Ducati (at least most things running the M3C ECU). 

I tested a Monster 696 by unplugging the MAP sensor (incase it was using that for sync) and looking at the trigger pattern. It starts off in wasted-spark mode:

[image]

Then after a few seconds goes into sequential mode:

[image]

How does it achieve sync with no cam sensor and no MAP pulse to reference? WHO KNOWS. My running theory is that since the 90 degree vtwin fires in a very non-linear fashion (cylinder 1 fires, 270 degrees later cylinder 2 fires, then a whole lot of nothing until its time for cylinder 1 to fire again), the ECU may be looking at the crank acceleration to determine when the cylinders are at the compression stroke and then switching to sequential mode.

TLDR: if you're happy with wasted-spark, then 24-2 missing-tooth trigger wheel and you're done.

The Testastretta Ducati timing gears have only one gap, so those are a 48-2 cam-speed wheel:

[image]

denandz, 2021-10-03:
FWIW I've just fired up a 696 engine using a 24-2 crank-speed decoder. The first-tooth angle was 210 degrees.

denandz, 2022-03-05:
Okay, after doing some more thinking and digging and logging, it looks like the way the stock ECU achieves sync is by analysing the crank acceleration. It fires up in wasted spark, then I assume its looking at the tooth data (or Siemens equivalent) in real time to figure out phasing. I pulled the trigger log of the engine running and graphed the results, you can see the acceleration difference between TDC exhaust and TDC compression.

[image]

I'm currently running a non-speeduino aftermarket ECU that supports crank-acceleration syncing on cranking, rather than firing up wasted-spark and then achieving sync. This works awesome, the bike fires up very quickly and I've had no issues. 

This acceleration-syncing technique could be used on single cylinder and uneven fire twins and allow for full-sequential operation with only a missing tooth crank sensor

ferraritobias, 2024-12-22:
> FWIW I've just fired up a 696 engine using a 24-2 crank-speed decoder. The first-tooth angle was 210 degrees BTDC cylinder 1 (the horizontal cylinder).

I'm installing a standalone injection in a Ducati scrambler with this same trigger pattern. 48-2-2. I'm going to configure the injection to work with 24-2, wasted spark, but I'm unsure about the timing teeth, would you know which tooth is equivalent to the 210 btdc?

denandz, 2024-12-23:
> I'm installing a standalone injection in a Ducati scrambler with this same trigger pattern. 48-2-2. I'm going to configure the injection to work with 24-2, wasted spark, but I'm unsure about the [...]

The first tooth after the -2 gap is 210 btdc. Double check with a timing light ofcourse!

ferraritobias, 2025-01-18:
> The first tooth after the -2 gap is 210 btdc. Double check with a timing light ofcourse!

The bike worked fine here with these settings, but now I need to make a more precise adjustment, 
Would you have or know of any volumetric efficiency map for a Ducati +- 800cc?
And as for the ignition advance when starting, do you have any idea of ​​a legal number to use?
I don't have much idea of ​​the flow rate of the original Scrambler 800 nozzles, that's also hard to find.

denandz, 2025-01-18:
Excellent! You could use the MaxxECU software and my Monster 696 base-file
as a starting point for VE numbers and ignition timing:
https://maxxecu.com/files/Basemaps/Engine/Ducati%20696.zip

denandz, 2025-01-18:
Sorry, this probably needs some more explanation. The 696 has a completely different intake system (ITBS) on top of the bore-stroke changes, the scrambler has a single throttle body so the VE numbers are going to be different. The 696 tune file should get you started, but I wouldn't be suprised if its pretty far out, especially in the low and mid range.

The top-end torque figures are the same, but the 803 scrambler engine makes more torque earlier in the RPM range.

Fuel injectors  If you can't find the flow rates for the Siemens Deka units and don't want to get the flow tested, you could use Bosch 381cc EV14 injectors. That's what I used in the 696 and they worked great.