speeduino/speeduino #420: DFCO changes TPS value!

jonbills, 2020-07-17:
I have 0.4.3c board, Alpha-N 6 cylinder batch/wasted spark setup with 36-1 trigger. 202005 firmware.
With DFCO disabled my TPS signal is pretty constant and if I lift off, TPS consistently reports 0.
when I turn DFCO on and lift off, TPS bounces around between 0 and 1,2,3. This makes for quite a disconcerting experience with the amount of engine braking varying as I approach a corner.
When I turn DFCO on, I have TPS Threshold at 1. I can't put it higher because I have quite big ITBs and anything above that might well be cruising condition. 
I've got a vague feeling I tried it in an earlier firmware and the problem wasn't there, but couldn't swear to it.
I've had a little browse around the code and didn't see anything obvious but then I'm unfamiliar with the code and the Arduino IDE so wouldn't read anything into that :)

runesmdk, 2020-07-29:
To get around the low load TPS signal, try using map when possible via the VE2. 

Set a conditional switch at around 92-95kpa, so when below and a vacuum signal is present and strong enough, it uses VE2 and map sensor as load signal. It gives you heaps of resolution in low load areas when running itb's.

jonbills, 2020-08-16:
> To get around the low load TPS signal, try using map when possible via the VE2.
>
> Set a conditional switch at around 92-95kpa, so when below and a vacuum signal is present and strong enough, it uses [...]

Thanks - so essentially, switch to MAP and only switch to TPS based load as it approaches zero vacuum? do you need to have MAP signal from all ITB runners joined, or is one good enough?

jonbills, 2022-07-09:
Problem went away with swap to SLC Free 2 wb02. weird.

closed 2022-07-09 by jonbills