speeduino/speeduino #421: Mitsubishi 3A92 Trigger

ElDominio, 2020-07-19:
I have a 3 cylinder Mitsubishi Mirage project car with a 3A92 engine, with this trigger wheel combo in the following pictures. I have some plans and would like trigger support to get it running ASAP!

[image]
[image]
[image]
[image]

jchurc16, 2020-08-19:
Can you advise what year your car is? And also what engine is fitted?

dascons, 2020-08-21:
Says in the OP:
I have a 3 cylinder Mitsubishi Mirage project car with a 3A92 engine

jchurc16, 2020-08-21:
Ah yes, sorry I had missed that. Just looking for a service manual with the trigger pattern now.

mike501, 2021-01-14:
I've hopefully created a trigger wheel for ardu-stim. Its in git as a pull request at the moment.

However what I've created is based on the photos, they don't appear to match the picture above (standard wave pattern). If you look at the cam in the picture, its got 4 teeth which are equally sized. The standard wave pattern has 4 gaps, 3 the same size and one smaller.  From the detail at the start of this post the cam generates a pulse in the gaps on the crank so this is what I created.

ElDominio, 2021-03-20:
I finally got the car back, and the ECU will be wired up this week, testing will begin soon! Where is the code at? @mike501

mike501, 2021-04-02:
Can we get a log of the triggers to prove if its as you described or if its as per the photo jchurc16 posted.

ElDominio, 2021-05-29:
[anotherlog.csv](https://github.com/noisymime/speeduino/files/6565086/anotherlog.csv)
[bestlog.csv](https://github.com/noisymime/speeduino/files/6565087/bestlog.csv)
[compositelognewcode.csv](https://github.com/noisymime/speeduino/files/6565088/compositelognewcode.csv)
[crankinglog.csv](https://github.com/noisymime/speeduino/files/6565089/crankinglog.csv)
[lookswaybetter.csv](https://github.com/noisymime/speeduino/files/6565090/lookswaybetter.csv)
[mrelogs3cyl.csv](https://github.com/noisymime/speeduino/files/6565091/mrelogs3cyl.csv)
[newestlogwithANOTHERUPDATE.csv](https://github.com/noisymime/speeduino/files/6565093/newestlogwithANOTHERUPDATE.csv)
[toothlog.csv](https://github.com/noisymime/speeduino/files/6565094/toothlog.csv)
[turnedon.csv](https://github.com/noisymime/speeduino/files/6565095/turnedon.csv)
[WHENITMAKESCOMPRESSION.csv](https://github.com/noisymime/speeduino/files/6565096/WHENITMAKESCOMPRESSION.csv)
[withnewcodelog.csv](https://github.com/noisymime/speeduino/files/6565097/withnewcodelog.csv)
[withnewcodetooth.csv](https://github.com/noisymime/speeduino/files/6565098/withnewcodetooth.csv)

Here's a ton of logs if anyone needs them, we're probably gonna be adding a 36-1 wheel externally and grinding three teeth off the cam

ElDominio, 2021-09-26:
i'm working on this trigger again, let's see if I can get it running this time

VitorBoss, 2021-11-07:
Just sent a PR with this decoder, need testing, RPM and sync working out of the box and sync almost immediately (hope so).

I can't login to bountysource with github or facebook accounts.

closed 2026-11-14 by ElDominio