speeduino/speeduino #429: Contributing tests (+ dead code removal)

FergusInLondon, 2020-08-06:
Loved the [recent development stream](https://www.youtube.com/watch?v=9sLUkkVAbnc) covering contributions and writing tests, @noisymime. I'm just wondering if there's any specific areas of the codebase where tests would be appreciated? I've poked around the `tests/` directory and can see roughly where there's already existing coverage.

As a side note, in areas like [`test_schedules.cpp`](https://github.com/noisymime/speeduino/blob/master/test/tests/test_schedules/test_schedules.cpp) I see some dead code commented out - I think I've seen it in the firmware itself too. Would there be any opposition to removing any of this that I stumble across whilst exploring the codebase?

Cheers

noisymime, 2020-08-09:
Commented out code that is retained is _usually_ things that are either experimental or not yet completed or working. I wouldn't generally consider them 'dead' code and removing them is probably not desirable. This isn't always the case, but is more common that code that is just plain not needed at all. 

The code in `test_schedules.cpp` for example is a series of tests that are wanted, but that are currently causing issues. Enabling them causes the arduino to crash. It's retained because its 90% complete and really only needs someone to debug it and figure out what is happening. 

Otherwise for tests it's really a matter of just picking a piece of code and start hammering away on it. For a given function, every combination of input, if statement result, possible return values etc needs a unit test to confirm it's performing the expected action.

adbancroft, 2025-08-26:
> Otherwise for tests it's really a matter of just picking a piece of code and start hammering away on it. For a given function, every combination of input, if statement result, possible return values etc needs a unit test to confirm it's performing the expected action.

The problem with this assumption is that the code base is **not** unit test friendly. I tend to find a lot of refactoring is needed to make a section of code unit testable. E.g. PR #1308. Maybe it's just me?