speeduino/speeduino #430: Data contention on toothLastToothTime and toothLastMinusOneToothTime

TBAMax, 2020-08-06:
No time to fix now, so just leave a note here about that so it do not get out of head:
There is read write contention issue between interrupt and main loop at least with the
"toothLastToothTime" and "toothLastMinusOneToothTime" global variables
Those variables are read using `timeToAngle` and used to determine dwell angle (and from this ignition angles), but interrupt can occur between reads of two variables and variables are edited in interrupt at every other engine rotation, so it will throw off ignition timing randomly.
At least need to verify this later that this is the case. 

BR
Teemo

TBAMax, 2020-08-06:
Ok sorry, I was looking at the wrong method in the `timeToAngle` there. Seems like the other method is not used. At least is not used in something important.

noisymime, 2020-08-06:
The only line that could be affected by this is https://github.com/noisymime/speeduino/blob/master/speeduino/crankMaths.ino#L63 (and the corresponding line for `angleToTime()`: https://github.com/noisymime/speeduino/blob/master/speeduino/crankMaths.ino#L32).

Eg:
`returnAngle = ( (unsigned long)(time * triggerToothAngle) / (toothLastToothTime - toothLastMinusOneToothTime) );`

So the risk here is only during the time it takes for the subtraction to occur, which about 8 clocks. It's a fairly low risk in the scheme of things, certainly much lower than for most calcs, but I agree that it's possible. 

Solution here is relatively simple, that subtraction needs to be pulled out to its own line and wrapped in `noInterrupts()`.

VitorBoss, 2020-08-06:
Isn't safer to use a copy of it before the math? Just use a `uintX_t lastToothTime = toothLastToothTime;
uintX_t toothTime = toothLastToothTime;`

That way is relatively easy to implement and safer too, it could even use the `noInterrupts()` before the copies reducing the delay to enable interrupts.

TBAMax, 2020-08-06:
I was looking at the current implementation of `doCrankSpeedCalcs()` in crankMaths.ino
And it seems to me that it would be logical to move its dutys to be with the every individual decoder.
doCrankSpeedCalcs() currently do the following things that I managed to notice, calculates following variables:
timePerDegreex16 - (adds one long divide to the main loop)
timePerDegree
degreesPeruSx2048- seems to be currently unused (adds long divide to the main loop)
degreesPeruSx32768- (adds long divide to the main loop)

All those seem to be better calculated with the decoder for every specific trigger pattern. Not in the interrupt part of the trigger pattern but same as RPM and crankangle is calculated with every decoder. To be exact it would be nice to calculate those with RPM as then all needed volatile variables are already freshly extracted.

noisymime, 2020-08-06:
Yeah you can pull them into temp variables (That would 100% have to be wrapped in `noInterrupts()`!) instead, either way would be fine and it's really just personal preference which way you go and I can't see any real performance difference

noisymime, 2020-08-06:
> All those seem to be better calculated with the decoder for every specific trigger pattern. Not in the interrupt part of the trigger pattern but same as RPM and crankangle is calculated with every [...]

There's no benefit to this that I can see. The data is just as 'fresh' when this is performed in `doCrankSpeedCalcs()` because it's all happening within the same loop. Everything within a single mainloop operates of a single point in time snapshot for it's calculations, so whether that is done in the `getRPM()` call or in the `doCrankSpeedCalcs()` call makes no difference. 

The downside to putting it in the `getRPM()` call is that you're reproducing the same piece of code for every single decoder when it is identical in every case.

TBAMax, 2020-08-07:
So I close the topic here for now. Topic discussed clear. No issues with that at the moment.

commit 9a6be330 (2020-08-08): Prevent interrupts when calculating tooth to tooth time (Resolves #430)
closed 2020-08-07 by TBAMax