speeduino/speeduino #439: [Feature Request] ALS - Antilag - Rotational idle

requies, 2020-08-18:
I'd like to request the addition of an ALS - Turbo Antilag system and rotational idle (fuel or spark cut).

The ALS system should be able to use the idle valve or an on/off or PWM output.
It would be nice to have some tables for the ALS timings and fuel add/remove. 

Thank you

bossthetuga, 2022-12-18:
I support this feature, it would be a great addition to speeduino boards for motorsports!

Eldonratzlaff, 2023-03-18:
Hello
What I saw in my experience is that antilag itself does not destroy engines. I saw in my experience two engines that destroyed the cylinder walls in a block, but both had in common that they used wasted spark, and all just look at what they are doing ignition 20 or 30 degrees after TDC but they don't see the waste spark that is 20 or 30 degrees in INLET. It is necessary to note that the use of antilag must be in a sequential spark configuration.