speeduino/speeduino #447: August Release, Teensy problems

kimdrummel, 2020-08-24:
Found some bugs in the new Firmware when using Teensy.

*When calibrating Custom #1 Temp Sensor in Tools, it seems to also alter the properties of the intake air temperature sensor.

*The MAP reading shows a value that is too low.
(I use the MPX6400 and even when calibrating for it, it shows about 66kpa, when it shows just under 100 on the may Firmware.)

*The TPS will not read any higher than 40 raw value.

Tried going back and forth between August and May release. And the problems with the TPS and MAP is only on the August release. Did new calibrations each time.

Attached Zip with Log when trying this out on Stim, and the tune used with it.
[Log+Tune.zip](https://github.com/noisymime/speeduino/files/5115940/Log%2BTune.zip)

noisymime, 2022-01-08:
Are you still seeing this? I can't reproduce this at all, switching between `202103`, `202108` and current master. MAP, TPS, CLT readings are all identical between them.

noisymime, 2024-03-15:
Closing as stale

closed 2024-03-15 by noisymime