speeduino/speeduino #459: VVT bugs in 202008

pazi88, 2020-09-30:
In 202008 release (and in current master) the VVT output gets stuck at 100% duty in open and closed loop modes. And on on/off mode the output gets stuck at 0% duty. The duty gauge in TS works as it should but the actual output signal for VVT solenoid gets stuck to that 0% or 100% duty.

I'm creating this issue, so that the bug doesn't get forgotten. I have done fix for the output getting stuck at 100%: https://github.com/noisymime/speeduino/compare/master...pazi88:VVT-fix

But can't seem to figure out why the output gets stuck at 0% duty in on/off mode. The output works few times from 0% duty to 100% duty and then gets stuck to 0%.

noisymime, 2020-12-01:
Just to check, is this still an issue in master? If so I'll look at it ASAP ahead of the next release

pazi88, 2020-12-02:
@noisymime Yes it is. There is also another, which is that if the missing tooth cam trigger event comes after 127 crank degrees, the value send to tunerstudio overflows. You don't notice that at all, but if you try to use closed loop vvt feature with overflowing cam degree value, the PID wont work at all.

I have fixed the overflow and stuck to 100% problems in PR #461 but the stuck at 0% wheb using open loop still remains mystery for me

pazi88, 2021-08-18:
This all is fixed in 202108

closed 2021-08-18 by pazi88