speeduino/speeduino #464: Programmable outputs with 2 conditions

sandmanWAjeep, 2020-10-20:
**What I wanted to happen:**

- have speeduino control the condenser fan and A/C compressor relay 

- When the A/C button in the car ( AUX Input 0) is pressed, the compressor relay ( programmable output 1, pin 6) will turn on followed by the condenser fan (programmable output 2, pin 43)

- TPS must be < 90% for output 1 to be on

- If CLT is >94*c, output 2 will be on regardless of Input 0

**I setup the programmable output rules as follows**

Select Rule Number = 1
Output Pin Num = 6
Activation Delay = 1
2nd Condition = AND
Condition 1 = Aux in 1 == 0
Condition 2 = TPS < 90

Select Rule Number = 2
Output Pin Num = 43
Activation Delay = 2
2nd Condition = OR
Condition 1 = Aux in 1 == 0
Condition 2 = coolant > 94

This did not achieve the desired outcome. I put some "serial prints" in the utilities.ino file to try and track down what was happening and it appeared as if the second condition for Rule Number 1 is being used when evaluating Rule Number 2.

I did achieve the desired outcome by setting up Rule Number 1 as a "dummy rule" and using rules 2 and 3 to control the outputs

Select Rule Number = 1
Output Pin Num = 44   (unused pin)
Activation Delay = 1
2nd Condition = AND
Condition 1 = Aux in 1 == 0
Condition 2 = TPS < 90

Select Rule Number = 2
Output Pin Num = 6
Activation Delay = 1
2nd Condition =  AND
Condition 1 = Aux in 1 == 0
Condition 2 = coolant < 90

Select Rule Number = 3
Output Pin Num = 43
Activation Delay = 2
2nd Condition =  OR
Condition 1 = Aux in 1 == 0
Condition 2 = VE1 > 94                                         (VE1 was chosen randomly. output 3 turns on when coolant is >94)

Side Note: It would be nice if the Aux input labels were the some on both the "Local Auxillary Input Channel Configuration" page and the "Programmable outputs" page. They are labeled 0 - 15 on the former and 1 - 16 on the latter.

elyrizal, 2020-10-21:
also it will not respond if selecting output polarity to active low. if select active low, output will always at high

bap168, 2020-10-22:
Editing the speeduino.ini fixed my issue too.

closed 2020-11-10 by sandmanWAjeep