speeduino/speeduino #480: Injector lag time - can be improved?

XPbIM3, 2020-11-23:
I have two things to ask about injector lag timing:

1)We have an opening lag time vs voltage calibration curve but have no handy tool to obtain correct values.  I do know two methods to obtain correct lag on working engine within diffirient voltages onboard:   
  - track an AFR miss (% correction required) when targeting different AFRs  on idle 
  - do a switchover between 1-2-4 pulses per cycle - resulting AFR should be somewhat close with correct lag time  - but speeduino requires a full restart when changing inj pulses per cycle. Also changing an injector time also requires full restart -   not so comfy. Can this be improved to have a tool to online calibrate lag time?

2) Injectors may have several others factors to consider when calculating lag time:   lag time vs voltage(lower V - harder to open - done),   lag time vs ramp pressure (higher MAP readout - more pressurized fuel in ramp - harder to open inj - more lag time),   some injectors may have "lag vs temperature" drift - mine do,  thats can be considered as  "lag time vs  fuel temp".
Also... it seems we have 0.1ms lag time precision in speeduino,  is this really enough?

pazi88, 2020-11-23:
0.1ms is plenty accurate because of the voltage correction. Lets say you have dead time of 0.101ms at 12v. Just put 0.1ms as dead time and 101% as voltage correction for 12v and that equals to 0.101ms

muuris, 2020-11-23:
Injector pressure does affect dead time, but it is assumed you input the values measured/know with your used fuel pressure. The pressure ratio across injector (fuel rail to manifold) should be constant, that's why the pressure regulator is for and no correction on dead time is required while running. This assumes of course you use MAP-referenced regulator.

The best way is to test injectors with different pulsewidths (with known voltage, pressure ratio and pulse count). This feature is currently not in firmware, but is quite easy to implement.

XPbIM3, 2020-11-23:
Pressure across injector is a constant - missed that moment ,  that's correct.

closed 2022-01-05 by noisymime