speeduino/speeduino #497: Miata trigger decoder not stable when applying trigger angle correction

Thommmm, 2021-01-04:
When setting a trigger angle correction when using the Miata trigger decoder, the ignition angle becomes very unstable. For instance, when fixing the output angle at, say, 10°, the output ignition angle is not actually fixed. It jumps all of the place and makes the engine not run. A fix is to set the angle correction to 0 and physically move the trigger device (which is possible). I have a Miata, so I log data if required.

noisymime, 2021-01-04:
This is something I suspected may have been a potential issue since the new ignition mode was added. Like most patterns, the decoder for this was designed to work with the pattern in its 'correct' form (ie with a trigger angle of 0) but it seems to be one where people take some liberties with the CAS being moved around. 

I'll get this addressed so that non-standard trigger angles can be used with the new ignition mode.

Thommmm, 2021-01-05:
Another option would be simply gray out the offset box in tunerstudio and fix it to 0 when using the Miata/4G63 trigger. I'm fine with having to move the trigger itself but was under the impression (so was another guy in the FB group) that we could just set the offset (and so we did).

lsdlsd88, 2021-01-07:
to clarify, which miata decder are you referring to? na or nb?

adbancroft, 2021-01-17:
> Another option would be simply gray out the offset box in tunerstudio and fix it to 0 when using the Miata/4G63 trigger

And the 99-05 trigger (`void triggerPri_Miata9905()` includes a check for zero trigger angle)

adbancroft, 2021-01-20:
> I'm fine with having to move the trigger itself but was under the impression (so was another guy in the FB group) that we could just set the offset (and so we did).

Same here. There's enough generic instructions on the web to "check timing & adjust" that this special case is easy to miss.