speeduino/speeduino #498: IAC stepper stepping forever

gdiciocco, 2021-01-09:
I observed this behaviour on a teensy 3.6 board, stepper open loop, minimum steps = 1, raising it makes the problem disappear.

Took a quick look at doStep() but been unable to find problems there.

Going to investigate further and on different MCU in the next days, opening the issue so other can confirm/cooperate.

Thank you,
Giacomo.

ric355, 2021-01-09:
As per my posting on Slack;

Looking at the master branch, the value for config6.iacStepHyster is always set to zero when the code is running, despite the value in the dialog being set to 1. One of the things the firmware does is, when the engine is off, it repeatedly calls "disableIdle()". That function keeps calling dostep() which in turn tries to take a step. If the hysteresis (min steps) is incorrectly set to zero the test for whether we need to make a step will always return true, even if current and target are already equal to each other. So that's why it keeps on stepping.

The root cause of the problem is that the speeduinoi.ini file has a 'translate' value of 1.0 in the config for iacStepsHyster. So in order to get a value of 1 you must specify a value of 2.  This change was made by @noisymime about 4 months ago.  I'm guessing it is just a mistake but he might want to chime in if there was a specific intent behind the change.

It looks like he was trying to set the minimum value to 1 but changed both the minimum and the offset. Most people won't see the problem as they run the default of 3 (but will have been getting an actual of 2 in the last release).

henriquebelaz, 2021-01-09:
If I'm not wrong that also happens in ATmega. Have set 2 steps as minimum and worked fine since.
I thought that was a stepper limitation. Nice to see it's SW related.

ric355, 2021-01-09:
> If I'm not wrong that also happens in ATmega. Have set 2 steps as minimum and worked fine since.
> I thought that was a stepper limitation. Nice to see it's SW related.

Correct; my debugging was on on a Mega.

noisymime, 2021-01-09:
This change was made in PR #390 from @celturbo. 

I'll have to go back through that and see if it was made for a specific reason, but on the face of it I can't see why that offset of 1 would be needed/wanted.

celturbo, 2021-01-10:
With a value of 0 in the Hysteresis check, it did not work correctly, which caused an increase in idleStepper.curIdleStep. With value with value> = 1, it does not happen.

https://github.com/noisymime/speeduino/blob/d5e4370343e938b1637b55694f8286cde60da1a1/speeduino/idle.ino#L572

PR #504 (2021-01-22): Restore the 'translate' attribute for iacStepHyster to  0
Fixes #498 by preventing the value sent from TS to the firmware from being zero when a value of 1 is selected in the dropdown. Note the original change set the minimum value to 1; this has been retained.
commit dd7c21ba (2021-01-22): Restore the 'translate' attribute for iacStepHyster to  1. Fixes #498 by (#504)

preventing the value sent from TS to the firmware from being zero when
a value of 1 is selected in the dropdown.    Note the original change
set the minimum value to 1; this has been retained.
closed 2021-01-22 by noisymime