speeduino/speeduino #499: External baro input doesn't work

TrinomCZ, 2021-01-09:
Hi,
during testing of black STM32F407VE6 board has been discovered, that external baro input doesn't work. When I tried to swap pins between various inputs, the problem moved with the baro pin, so it's clearly a SW issue.
External baro input has been enabled and input pin in TunerStudio has been always selected accordingly to what I had in init.ino file (there is no "board default" in this drop-down menu).
This issue seems to be present in all SW versions including the latest one (2020-12).

noisymime, 2021-01-09:
What is your hardware setup for this? I have many boards here with external bar sensors that are working without issues using a 2560, so I would be surprised if it's an issue in the firmware, but it maybe something specific to your hardware setup.

TrinomCZ, 2021-01-10:
Hi, I have just bare black STM32 development board with a potentiometer connected to whatever input pin I need to test.
[image]
In Speeduino file init.ino, it uses case 60, where on line 2075 is `pinBaro = PB1;`. As I mentioned in the first post, if I swap this to let's say PC1 and what used to be assigned to PC1 is now assigned to PB1, the problem moves around. At this moment, everything is back to default.
In tunerstudio, I can clearly see, that even though I have set baro input to external source, it still ignores these settings and baro value just jumps around MAP value (no, it definitely isn't a floating average of MAP).
[image]
[image]
At this moment, there is no RPM input and baro input pin is tied to ground.

noisymime, 2021-01-10:
You've got `PB1` selected in tunerStudio as the pin, but that isn't one of the options in the ini file. If you've modified that file, have you verified that the entry you've got there for `PB1` matches the actual pin number? 

You can't just add `PB1` to the list in any place, its location in the array represents the pin number that will be used.

TrinomCZ, 2021-01-10:
Oh, I haven't realised, that I copied that whole line from an older ini file, because it lacks the "board default" option. I clearly recall, that using any of the "Axx" bricks my STM board and I had to manually erase the external flash in the past :(
I've put the line with Axx back in the ini file, but that doesn't solve my issue, so what would you recommend? Could you please point me out, where this input is being processed in the speeduino ino files to help me understand, which input should I use or how to force it to use "board default" setting, which is defined in init.ino?
Thank you.

TrinomCZ, 2021-01-21:
Any updates or ideas?

VitorBoss, 2021-01-21:
First, the arduino way for read analog channels listed on Tuner Studio won't work.
Second, the pin numbers must follow the library standard listed at https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/BLACK_F407XX/variant.cpp#L49

Please keep the USB D lines out of listing, selecting them shouldn't hang the code anymore, but is safer to keep them out

TrinomCZ, 2021-01-21:
Victor, could you please try to explain it more in detail? Why every single analog input works, except for the baro input? If you say, it doesn't work in tunerstudio, how is the pin, which I can select in Tunerstudio transmitted to Speeduino and then translated into analog inputs? I'm asking, because I didn't fully understand the code, however selecting any of the Axx inputs causes STM board to brick. To unbrick it, I have to manually delete whole flash memory, otherwise it won't work (the port addressing is clearly wrong).
From my point of view (which might be wrong), I need to either force Speeduino to use the predefined input and totally ignore any settings comming from Tunerstudio or send the correct input from the Tunerstudio in the first place (which will be accepted by Speeduino).

In the past, I discussed this in the STM32 development thread and I saw someone modifying the speeduino.ini file for TS to include the PB0, PB1,... along the original A0, A1,... and they claimed, it works for them, so I did the same. Also this is the only way, how I was able not to brick my board, when I turned on the external baro input, though it didn't work.

TrinomCZ, 2021-02-09:
I see no response here, so I tried putting everything back to default (including the ini file for Tuner Studio) and flashed the chip again. I tried to select the A9 option, because that seems to be the right one for PB1 of my STM32 board (according to "board_stm32_generic,h"). This time, selecting the A9 didn't brick the board (heureka!), but also didn't do anything. I tried other inputs, but without any success. Then I've discovered, that it completly ignores all defined analog inputs. I've checked that in Tuner Studio with correctly defined AUX0 (to analog A0) and AUX1 (to analog A1 or others) and visible gauges on the dash, which don't move at all, when I apply voltage to PA0 and PA1 (or others accordingly).

So that might be the main problem.

VitorBoss, 2021-02-09:
I'll try to explain better this time, on ini there is a list of pins from 3 to 53, anything above that is treated by speeduino as analog channels because Mega2560 only have 53 IO pins and 16 analog ones, for STM32 it isn't true, for F407 there is 73 IO pins including 16 analog ones and the code is absolutely the same, for [F407 PB1](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/BLACK_F407XX/variant.h#L43) is pin 57 and isn't listed on the INI.

Any particular reason why you can't use the default pin for baro?

TrinomCZ, 2021-02-09:
Ah, I see. So the right input for PB1 should be A13. And I can confirm, that it works.
Thank you very much!

closed 2021-02-10 by TrinomCZ