speeduino/speeduino #510: Unable to repurpose unused function pins as programmable outputs

aaronjonesxyz, 2021-01-24:
A commit on Dec 2 to utilities.ino added code to the programmable output functions that checks if the selected output pin for a programmable output is already reserved for another function, but doesn't seem to also check if that function is enabled. 
This seems to disable the ability to repurpose the high/medium current drivers for unused functions like VVT and Idle as programmable outputs.

VitorBoss, 2021-01-24:
Yes, that was expected, the next step is to add a check to pin function is enabled like this:

`#define pinIsOutput(pin)    ( ((pin) == pinFuelPump) || (((pin) == pinFan) && (configPage6.fanEnable == 1) || (((pin) == pinVVT_1) && (configPage6.vvtEnabled > 0) || ((pin) == pinVVT_2) && (configPage6.vvtEnabled > 0) || (((pin) == pinBoost) && (configPage6.boostEnabled==1) || (((pin) == pinIdle1) && ((configPage6.iacAlgorithm==0) || (configPage6.iacAlgorithm>=4)) || (((pin) == pinIdle2) &&(configPage6.iacChannels == 1)) || ((pin) == pinTachOut) )

#define pinIsInjector(pin)  ( ((pin) == pinInjector1) || (((pin) == pinInjector2) && (configPage2.nInjectors>1)) || (((pin) == pinInjector3) && (configPage2.nInjectors>2)) || (((pin) == pinInjector4) && (configPage2.nInjectors>3)) || (((pin) == pinInjector5) && (configPage2.nInjectors>4)) || (((pin) == pinInjector6) && (configPage2.nInjectors>5)) || (((pin) == pinInjector7) && (configPage2.nInjectors>6)) || (((pin) == pinInjector8) && (configPage2.nInjectors>7)) )
`

aaronjonesxyz, 2021-01-24:
Perfect, finding the function enable variables and adding them to the check was what I was planning on doing too!

closed 2023-11-30 by aaronjonesxyz