speeduino/speeduino #546: Failing unit test

adbancroft, 2021-03-21:
2 initialization tests fail:

- `test\test_misc\tests_misc.cpp:51:test_initialisation_outputs_V03:FAIL: Expected 1 Was 0. Idle 2 [FAILED]`
- `test\test_misc\tests_misc.cpp:80:test_initialisation_outputs_MX5_8995:FAIL: Expected 1 Was 0. Idle 2    [FAILED]`

Traced back to fd9fa36c2968ff0bf673e150bb2cea6fa3a4dbb3

noisymime, 2021-03-22:
What platform are you running the tests on? These are currently passing on mega2560 and I will test shortly on Teensy35

adbancroft, 2021-03-22:
```
PS C:\dev\speeduino> pio test -f "test_misc"
Verbose mode can be enabled via `-v, --verbose` option
Collected 3 items

Processing test_misc in megaatmega2560 environment
-------------------------------------------------------------------------------------------------------------------------
Building...
In file included from C:/dev/speeduino/speeduino/speeduino.ino:32:0:
speeduino\idle.h:72:20: warning: 'void idleInterrupt()' declared 'static' but never defined [-Wunused-function]        
 static inline void idleInterrupt();
                    ^~~~~~~~~~~~~
In file included from C:/dev/speeduino/speeduino/speeduino.ino:33:0:
speeduino\auxiliaries.h:83:20: warning: 'void boostInterrupt()' declared 'static' but never defined [-Wunused-function]
 static inline void boostInterrupt();
                    ^~~~~~~~~~~~~~
speeduino\auxiliaries.h:84:20: warning: 'void vvtInterrupt()' declared 'static' but never defined [-Wunused-function]  
 static inline void vvtInterrupt();
                    ^~~~~~~~~~~~
C:/dev/speeduino/speeduino/idle.ino: In function 'void initialiseIdle()':
C:/dev/speeduino/speeduino/idle.ino:78:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
       iacPWMTable.axisX = configPage6.iacBins;
       ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
C:/dev/speeduino/speeduino/idle.ino:80:5: note: here
     case IAC_ALGORITHM_PWM_CL:
     ^~~~
Uploading...
In file included from C:/dev/speeduino/speeduino/speeduino.ino:32:0:
speeduino\idle.h:72:20: warning: 'void idleInterrupt()' declared 'static' but never defined [-Wunused-function]
 static inline void idleInterrupt();
                    ^~~~~~~~~~~~~
In file included from C:/dev/speeduino/speeduino/speeduino.ino:33:0:
speeduino\auxiliaries.h:83:20: warning: 'void boostInterrupt()' declared 'static' but never defined [-Wunused-function]  
 static inline void boostInterrupt();
                    ^~~~~~~~~~~~~~
speeduino\auxiliaries.h:84:20: warning: 'void vvtInterrupt()' declared 'static' but never defined [-Wunused-function]    
 static inline void vvtInterrupt();
                    ^~~~~~~~~~~~
C:/dev/speeduino/speeduino/idle.ino: In function 'void initialiseIdle()':
C:/dev/speeduino/speeduino/idle.ino:78:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
       iacPWMTable.axisX = configPage6.iacBins;
       ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
C:/dev/speeduino/speeduino/idle.ino:80:5: note: here
     case IAC_ALGORITHM_PWM_CL:
     ^~~~

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file ".pio\build\megaatmega2560\firmware.hex"
avrdude: writing flash (116334 bytes):

Writing | ################################################## | 100% 18.64s

avrdude: 116334 bytes of flash written
avrdude: verifying flash memory against .pio\build\megaatmega2560\firmware.hex:
avrdude: load data flash data from input file .pio\build\megaatmega2560\firmware.hex:
avrdude: input file .pio\build\megaatmega2560\firmware.hex contains 116334 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 14.91s

avrdude: verifying ...
avrdude: 116334 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

Testing...
If you don't see any output for the first 10 secs, please reset board (press reset button)

test\test_misc\tests_misc.cpp:9:test_initialisation_complete    [PASSED]
test\test_misc\tests_misc.cpp:10:test_initialisation_ports      [PASSED]
test\test_misc\tests_misc.cpp:51:test_initialisation_outputs_V03:FAIL: Expected 1 Was 0. Idle 2 [FAILED]
test\test_misc\tests_misc.cpp:80:test_initialisation_outputs_MX5_8995:FAIL: Expected 1 Was 0. Idle 2    [FAILED]
test\test_misc\tests_misc.cpp:84:test_corrections_WUE_active    [PASSED]
test\test_misc\tests_misc.cpp:85:test_corrections_WUE_inactive  [PASSED]
test\test_misc\tests_misc.cpp:86:test_corrections_WUE_active_value      [PASSED]
test\test_misc\tests_misc.cpp:87:test_corrections_WUE_inactive_value    [PASSED]
test\test_misc\tests_misc.cpp:183:test_corrections_dfco_on      [PASSED]
test\test_misc\tests_misc.cpp:184:test_corrections_dfco_off_RPM [PASSED]
test\test_misc\tests_misc.cpp:185:test_corrections_dfco_off_TPS [PASSED]
test\test_misc\tests_misc.cpp:186:test_corrections_dfco_off_delay       [PASSED]
test\test_misc\tests_misc.cpp:10:test_PW_No_Multiply    [PASSED]
test\test_misc\tests_misc.cpp:11:test_PW_MAP_Multiply   [PASSED]
test\test_misc\tests_misc.cpp:12:test_PW_MAP_Multiply_Compatibility     [PASSED]
test\test_misc\tests_misc.cpp:13:test_PW_AFR_Multiply   [PASSED]
test\test_misc\tests_misc.cpp:14:test_PW_Large_Correction       [PASSED]
test\test_misc\tests_misc.cpp:15:test_PW_Very_Large_Correction  [PASSED]
test\test_misc\tests_misc.cpp:76:test_tableLookup_50pct [PASSED]
test\test_misc\tests_misc.cpp:77:test_tableLookup_exact1Axis    [PASSED]
test\test_misc\tests_misc.cpp:78:test_tableLookup_exact2Axis    [PASSED]
test\test_misc\tests_misc.cpp:79:test_tableLookup_overMaxX      [PASSED]
test\test_misc\tests_misc.cpp:80:test_tableLookup_overMaxY      [PASSED]
test\test_misc\tests_misc.cpp:81:test_tableLookup_underMinX     [PASSED]
test\test_misc\tests_misc.cpp:82:test_tableLookup_underMinY     [PASSED]
-----------------------
25 Tests 2 Failures 0 Ignored
============================================== [FAILED] Took 49.54 seconds ==============================================

Test       Environment     Status    Duration
---------  --------------  --------  ------------
test_misc  megaatmega2560  FAILED    00:00:49.544
========================================= 1 failed, 0 succeeded in 00:00:49.544 =========================================
```

adbancroft, 2021-03-22:
Mega2560.

Complete test run output above.

noisymime, 2021-03-22:
I'll have a look into that file, can you confirm you haven't modified or at all? 

This is passing on my system and on the automated testing (https://travis-ci.org/github/noisymime/speeduino/builds/763490211#L2072) so it could be a set of specific conditions on your end

adbancroft, 2021-03-22:
I'm running off of master, no mods.

If it's running on the CI system, it's probably something on my end. I'll do further investigation & let you know what I find.

noisymime, 2021-03-22:
I did just notice one of the warnings in the build that maybe related and have pushed a fix. It's possible that this test passes/fails based on the values in the test boards EEPROM, although this would be surprising.

Test with the latest commit and see if it makes any difference

adbancroft, 2021-03-23:
Still failing. Leave it with me.

noisymime, 2022-01-06:
Closing this one as the test seems to be passing on the automated builds. Can reopen if needed

closed 2022-01-06 by noisymime