speeduino/speeduino #555: Mitsubishi 4G94 Trigger (modified 4G63)

ElDominio, 2021-03-31:
Hey guys, me again

I've been trying to wrap my head around this trigger for the better part of two weeks, but seeing as I haven't been able to code a trigger successfully yet, here I am posting this.

I will include some code I wrote which syncs up ONLY with lite filter activated, and without New Ignition Mode active. While it syncs and apparently reads correctly, once I install spark plugs in the engine, nothing ever works again lol.

Code:
`/* -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Name: Mitsubishi  / NA/NB Miata + MX-5 / 4/2
Desc: TBA
Note: raw.githubusercontent.com/noisymime/speeduino/master/reference/wiki/decoders/4g63_trace.png
Tooth #1 is defined as the next crank tooth after the crank signal is HIGH when the cam signal is falling.
Tooth number one is at 355* ATDC
*/
void triggerSetup_4G63()
{
  triggerToothAngle = 180; //The number of degrees that passes from tooth to tooth (primary)
  toothCurrentCount = 99; //Fake tooth count represents no sync
  secondDerivEnabled = false;
  decoderIsSequential = true;
  decoderHasFixedCrankingTiming = true;
  triggerToothAngleIsCorrect = true;
  MAX_STALL_TIME = 366667UL; //Minimum 50rpm based on the 110 degree tooth spacing
  if(initialisationComplete == false) { toothLastToothTime = micros(); } //Set a startup value here to avoid filter errors when starting. This MUST have the initi check to prevent the fuel pump just staying on all the time
  //decoderIsLowRes = true;

  //Note that these angles are for every rising and falling edge
  if(configPage2.nCylinders == 6)
  {
    //New values below
    toothAngles[0] = 715; //Rising edge of tooth #1
    toothAngles[1] = 45;  //Falling edge of tooth #1
    toothAngles[2] = 115; //Rising edge of tooth #2
    toothAngles[3] = 165; //Falling edge of tooth #2
    toothAngles[4] = 235; //Rising edge of tooth #3
    toothAngles[5] = 285; //Falling edge of tooth #3

    toothAngles[6] = 355; //Rising edge of tooth #4
    toothAngles[7] = 405; //Falling edge of tooth #4
    toothAngles[8] = 475; //Rising edge of tooth #5
    toothAngles[9] = 525; //Falling edge of tooth $5
    toothAngles[10] = 595; //Rising edge of tooth #6
    toothAngles[11] = 645; //Falling edge of tooth #6

    triggerActualTeeth = 12; //Both sides of all teeth over 720 degrees
  }
  else
  {
    // 70 / 110 for 4 cylinder
    toothAngles[0] = 685; //Falling edge of tooth #1
    toothAngles[1] = 75; //Rising edge of tooth #2
    toothAngles[2] = 145; //Falling edge of tooth #2
    toothAngles[3] = 255; //Rising edge of tooth #1

    toothAngles[4] = 325; //Falling edge of tooth #1
    toothAngles[5] = 435; //Rising edge of tooth #2
    toothAngles[6] = 505; //Falling edge of tooth #2
    toothAngles[7] = 615; //Rising edge of tooth #1

    triggerActualTeeth = 8;
  }

  triggerFilterTime = 1500; //10000 rpm, assuming we're triggering on both edges off the crank tooth.
  triggerSecFilterTime = (int)(1000000 / (MAX_RPM / 60 * 2)) / 2; //Same as above, but fixed at 2 teeth on the secondary input and divided by 2 (for cam speed)
  triggerSecFilterTime_duration = 4000;
  secondaryLastToothTime = 0;
}

void triggerPri_4G63()
{
[... 366 lines cut ...]
      ignition3EndTooth = 4;
      ignition4EndTooth = 6;
    }
    else
    {
      ignition1EndTooth = 6;
      ignition2EndTooth = 2;
      ignition3EndTooth = 4;
      ignition4EndTooth = 2; //Not used
    }
  }

  lastToothCalcAdvance = currentStatus.advance;
}
`

Here's a trace using composite logger:
[image]

And here are some tooth and composite logs:

[toothlogs.zip](https://github.com/noisymime/speeduino/files/6234682/toothlogs.zip)

As always this will have a bounty attached. 
Thanks for the help!

sisco0, 2021-04-02:
You could get a better idea of what did you modify by taking a look at the files changed I created for the code you provide, please take a look at https://github.com/sisco0/speeduino/commit/4d66914922d6061af677396d69ee97864d0e4720.

I could summarize the tasks you got to do:
1. You modified `toothAngles` array.
2. You added some debugging information (`Serial.println` and `Serial.print`).
3. You removed `triggerFilterTime` variable.
4. You modified `triggerToothAngle`.
5. You modified some conditions, replacing `true` by `false`.

The modified conditions are like `READ_PRI_TRIGGER`, these are inverted for some reason.
There is some part of the code that talks about _noise_. Is this relevant for us? You modified the number there as well.

Please take care of reading the diff at the commit referenced before and see if it makes sense for your purpose.
I would be happy to help you.

ElDominio, 2021-04-02:
The primary triggers are inverted, as seen on the provided trace, that's why I flipped them.

I included the code sto see if it was useful to anyone wanting to code for it. Filters were removed so I could see what the crank was actually doing.

sisco0, 2021-04-02:
@ElDominio Could you send me a mail to siscomagma at gmail dot com so we could arrange a call for getting more information about the current problem had? I would be happy to contribute.

nuosu, 2021-04-11:
[image]

mike501, 2021-04-11:
Fab thanks. That's all I need to update my code that's working in principle against a guessed pattern.

mike501, 2021-04-13:
I've updated ardustim to do this pattern. I've been working on too many patterns at the same time so my source code control has got a bit lax and this also includes 3a92 fix, 4b11 and 4g94. (as well as a random Subaru pattern!)

https://github.com/mike501/Ardu-Stim/tree/4g94

Also due to working on too many patterns i'm confusing 4g94 and 4b11. Its 4b11 i have code working in principle. I'll look at 4g11 over the next few days. Looking at the pattern I'll aim to reuse 4g63 but put a different cam trigger pattern to trigger on the second smaller tooth (so always on the second cycle).

mike501, 2021-04-18:
Turns out you can't use 4g63 as is due to it having a tight coupling to the cam in its original form. New version of crank written and debug in process (it works but I'm not happy with ignition timing accuracy currently)

ElDominio, 2021-05-29:
It's also slightly different angles. 

We decided to adapt a 4g63 trigger on this car to make it run.

ElDominio, 2021-09-13:
Hey guys, I did the 4g93 trigger, which is also different from this.

I'm attaching my modified decoder and init, secondary trigger need to be changed to rising.
[image]
[4g93triggerfiles.zip](https://github.com/noisymime/speeduino/files/7157828/4g93triggerfiles.zip)

Keep in mind, I overwrote the 4G63 trigger to make it work, so I can't just simple merge it lol
I'll try to work on 4G94 next

josielbairros, 2021-12-18:
first i apologize for my english .  but would you like to understand is the 4g63 decoder different from the 4g64?  because I use speduino in an engine of my Pajero shogun (Tr4 in Brazil) 2.0 engine 4g94 with the 4g93 decoder, and the start is really unstable sometimes the engine does not start making it necessary to turn the ignition on and off so that the engine can set the cycle.  Could this be because I'm using the wrong decoder or is it some error coming from the 4g63 decoder?  thank you so much.

eduardocerqueirasilva, 2022-11-09:
eldomino is your fork working this decoder? i have a lancer 4g92 sohc that probably uses this decoder