speeduino/speeduino #559: Programmable output and glitch on RPM

Zeiberstein, 2021-04-19:
**Situation**
Speeduino 2021.03, Arduino Mega 2560, V 0.4 board, on a former carburetor with breaker points kitcar. 

**What do I want to establish?**
I use a programmable output for powering an lambda sensor after engine start. I use ‘RMP > 400’ as condition.

**What happened?**
I found out that, while the engine is running, this output isn’t stable. There occur short glitches where the output falls to zero. This happens several times per minute. 

**How to reproduce?**
Define a programmable output like this: 
[image]

Start the engine, start a log in TunerStudio and let it run idle for a few minutes.
Open the log in MegaLogViewer and display RPM (white) and output rpmGT400 (red):
[image]

**My 2 cents:**
When I add an extra programmable output with condition ‘(RMP > 400) OR (RMP > 400)’ the glitches don’t occur (displayed as a straight green line in the top of the graph)

[image]

[image]

I can only explain all above by assuming that checkProgrammableIO() is not threadsafe and sometimes ‘sees’ a wrong and a too low value for RPM.

neovin87, 2021-05-16:
Hi, I have found issues with the current ini, storage ino and h referencing different locations. 
I'm no firmware dev and just a bit of a hacker but using the eeprom read sketch in arduino as a debug tool the changes made to the programmable output page in tunerstudio are being wrote to the wrong locations and other information is being written there.

Can anyone else help here as I am doing it ass about face and need some secondary eyes

eqTom, 2021-07-18:
Do you see any loss of sync around the same time?

Zeiberstein, 2021-07-19:
> Do you see any loss of sync around the same time?
No, I didn't notice.