speeduino/speeduino #563: ESP32 adaptador bord

Carlos1Eduardo1Colpo, 2021-04-23:
with the use of shift registers Perhaps it would be possible for slower IO, add more pins for expansion, And add bluetooth and wifi compatibility.

pazi88, 2021-04-23:
If you mean to use esp32 as main Speeduino processor, then forget it. ESP32 has only 4 hardware timers, which is way too little for Speeduino usage. Also esp32 does panic reset every time something goes even slightly wrong, which is no go if you plan to control something like engine.

Carlos1Eduardo1Colpo, 2021-04-23:
I found this code below that causes ESP32 not to reset itself but if not used correctly it can cause errors or instability but maybe it is the solution for it to work without showing risks to the engine

#include "soc/rtc_wdt.h"
#include "esp_int_wdt.h"
#include "esp_task_wdt.h

rtc_wdt_protect_off();
rtc_wdt_disable();
disableCore0WDT();
disableCore1WDT();
disableLoopWDT();

WDT = Watchdog timer

muuris, 2021-04-24:
If wifi and bluetooth is what you want to add, then why not just integrating those into exististing MCU design? Bluetooth with HC05/06 works fine as is. ESP32 is good for making a web interface but as mentioned earlier, forget about using it as the main processor.

Carlos1Eduardo1Colpo, 2021-04-24:
so maybe a web enterface that can upload The tune or modify without the nasity cables or maybe a web page with instant data

closed 2021-04-25 by Carlos1Eduardo1Colpo