speeduino/speeduino #567: Teensy 3.5 31 pin Voltage Problem

sempoonet, 2021-04-25:
Hello, I have a Speeduino powered Scg-ecu 1.0 board. When I install the Speeduino v202103 version, there is 2.4 volts continuously on pin 31. This pin is the ignition pin. When I install the Speeduino v20208 version, 2.4 volts is not always available, but when I test it in Tunerstudio program, it comes to 2.4 volts on pin 31. 3.2 volts comes standard on other ignition pins.

opelpanfan, 2021-06-17:
Check your tune file if you have any functions assigned to pin 31. Base map has n2o_stage2_pin set to 31. As a start I suggest go through all unused functions and set all pins to max pin number for the teensy board (let's say 58). 
Also I remember that some issues where with 
#define BOARD_DIGITAL_GPIO_PINS 34
 #define BOARD_NR_GPIO_PINS 34

I had to set mine to 64 as teensy has 64 IO pins.

Messymechanic, 2021-07-21:
Sorry to comment on a few month old problem. But I just wanted to say thanks @opelpanfan . I've had a teensy 3.5 laying on the bench for months because of this issue and your advice fixed it,

KenaiFrankenstein, 2022-02-21:
> Check your tune file if you have any functions assigned to pin 31. Base map has n2o_stage2_pin set to 31. As a start I suggest go through all unused functions and set all pins to max pin number for the teensy board (let's say 58). Also I remember that some issues where with #define BOARD_DIGITAL_GPIO_PINS 34 #define BOARD_NR_GPIO_PINS 34
> 
> in board_teensy35.h file.
> 
> I had to set mine to 64 as teensy has 64 IO pins.

Thank you opelpanfan!, this also fixed my issue with teensy!!

commit 3c8726ff (2024-05-14): Correct IO numbers for Teensy boards. Fixes #567
commit 99085e9a (2024-05-14): Correct IO numbers for Teensy boards. Fixes #567
closed 2024-05-14 by noisymime