speeduino/speeduino #569: array fsIntIndex is declared incorrectly

Zeiberstein, 2021-04-27:
The array fsIntIndex contains all the indexes in currentStatus that should be considered as a double byte value. Indexes not present in this array are considered single byte values. For comparison in a rule for a programmable output, the corresponding value in fsIntIndex has to be known in order to determine whether a status parameter is Single cq. Double Byte.

I noticed that the byte array fsIntIndex is defined as:

`const byte PROGMEM fsIntIndex[31] = {4, 14, 25, 27, 32, 41, 43, 45, 47, 49, 51, 53, 55, 57, 59, 61, 63, 65, 67, 69, 71, 75, 77, 79, 81, 85, 87, 89, 96, 101}; `

This array has a defined length of 31, but only 30 elements were given. 
The missing 31th element is added as a zero by the compiler, so effectively the array is declared as:

`const byte PROGMEM fsIntIndex[31] = {4, 14, 25, 27, 32, 41, 43, 45, 47, 49, 51, 53, 55, 57, 59, 61, 63, 65, 67, 69, 71, 75, 77, 79, 81, 85, 87, 89, 96, 101, 0}; `

Because of the unwanted presence of 0 in this fsIntIndex, the status variable with index 0 will now erroneously be classified as a Double Byte (this happens to be status.sec1).

Note: I believe there are other problems with fsIntIndex (see https://speeduino.com/forum/viewtopic.php?f=13&t=4607), but I have no independent confirmation on that.

adbancroft, 2021-05-20:
2 other items:

1. The array is declared `PROGMEM`, so is stored in flash. Doesn't it need to be read using `pgm_read_word`?
2. It's declared `extern `in `globals.h`, but only used in `utilities.ino`. So just declare & define in one line in `utilities.ino`.

pazi88, 2021-05-20:
> 2 other items:
> 
> 1. The array is declared `PROGMEM`, so is stored in flash. Doesn't it need to be read using `pgm_read_word`?
> 2. It's declared `extern `in `globals.h`, but only used in `utilities.ino`. So just declare & define in one line in `utilities.ino`.

`pgm_read_byte?` at least works and there is no more issues with 2-byte variables with programmable outputs.

Zeiberstein, 2021-05-20:
> 2 other items:
> 
> 1. The array is declared `PROGMEM`, so is stored in flash. Doesn't it need to be read using `pgm_read_word`?
> 2. It's declared `extern `in `globals.h`, but only used in `utilities.ino`. So just declare & define in one line in `utilities.ino`.

Yes, I saw in Slack that pazi88, vitor_boss and noisymime are aware of the issue mentioned in https://speeduino.com/forum/viewtopic.php?p=49882#p49882. I'm confident they will address it. :-)

adbancroft, 2021-05-20:
> `pgm_read_byte` at least works and there is no more issues with 2-byte variables with programmable outputs.

Copy paste error - I meant `pgm_read_byte`!

commit 76f6a7f1 (2022-01-05): Use common (and faster) is2ByteEntry() function for programmable IO. This fixes #569 (Though it was likely resolved earlier anyway)
closed 2022-01-05 by noisymime