speeduino/speeduino #580: [Feature Request] Check Engine Light (CEL) Control

gadgethm, 2021-05-12:
I know this has been mentioned in previous pull requests, feature requests, and forum threads, but I didn't see a specific issue tracker open for this feature, so I figured I would open one.

It would be awesome to have some kind of CEL feature.  Some scenarios I can think of that could trigger a CEL turn on and latching:

- Sensors shorted/open or out of range
- Consistent sync loss (Not sure how to determine, maybe a number of sync losses in a time frame?  User configurable?)
- Low oil/fuel pressure (if sensors are configured right).  
- Light on power on if engine is not running to show the CEL is still working
- Generic limits that one can add; for example if boost pressure exceeds a certain value etc.

Thoughts?

borathtl, 2021-05-13:
This would be great addon
planning to work on this this summer if other projects allows me

shiznit304, 2021-06-20:
I currently use programmable output to control my CEL to tell me when coolant temp is grearter than 220 F. I would be neat to be able to make it flash at different rates to mean different things.

cgh459, 2021-10-01:
> I currently use programmable output to control my CEL to tell me when coolant temp is grearter than 220 F. I would be neat to be able to make it flash at different rates to mean different things.

Can you share how you did this?  How do you have your wiring set up and did you have to change any jumpers in the fishdog ua4c box?

shiznit304, 2021-10-02:
@cgh459 https://www.facebook.com/groups/311438776303488/posts/990955728351786/

Let me know if you can't see it and I'll try copy pasting my post and photos here.

cgh459, 2021-10-02:
> @cgh459 https://www.facebook.com/groups/311438776303488/posts/990955728351786/
> 
> Let me know if you can't see it and I'll try copy pasting my post and photos here. The CEL jumper is included in the post photos

Thats absolutely what I needed. Thank you.  Also another question. Do you need to physically open the fishdog ecu case and change any jumpers on the speeduino or is everything done on the adapter harness and in software?

sashan13, 2023-07-05:
Would be nice to flash it on each sync loss.

bahram-azimbeyk, 2026-09-10:
For a small Mega2560-compatible first step, would maintainers prefer a diagnostics-only extension using existing telemetry before adding a latched CEL/event system?

Speeduino already exposes RPM/boost/oil/AFR/coolant protection indicators in the INI. #150 also records existing temperature-calibration endpoint fallbacks, so neither basic protection indication nor all sensor fallback behavior should be treated as missing. The useful gap to define is how an actual sensor fault is qualified and distinguished from a protection condition or a normal cut.

Questions before implementation:

1. Which fault conditions should be included first, and what debounce/recovery/latching behavior is intended?
2. Should the first version only report faults, or also change controller behavior? Controller-specific fail-safe actions need separate agreement and tests.
3. Can existing output-channel/status information be reused, or is a versioned field needed? Where should any new configuration live without disturbing old tunes?
4. How should this align with the existing CEL PR #1096 and pin-priority work in #1604?

No new telemetry fields, EEPROM allocation or firmware behavior has been introduced. A bounded implementation with measured AVR costs would be preferable to a broad diagnostics framework; bench tests should distinguish sensor faults from normal operating conditions before assigning a CEL policy.