speeduino/speeduino #594: Feature request 2nd AFR table

jeagan626, 2021-06-18:
It would be nice to have a second AFR table that works the same way as the second VE table so one could switch between calibrations on the fly, or run a leaner tune during cruising conditions when say on flex fuel.

ThresherBuilt, 2021-08-22:
Not a bad idea. AFR table for 93 and one for E85. My targets are definitely different on e85 than pump gas. I leave it set up for my e85 targets since I use e85 99% of the time but a switchable table for that would be nice for datalogs and such

muuris, 2021-08-23:
I've also thought about 2nd AFR table, for the same reason (E85 vs pump gas targets differ). Switching option for other use cases, but for flexfuel use a blended (interpolated) value between the two tables could be used, with additional 2D table to set eth% / table2% (basically using the the one as in current flex settings dialog for this).

521Stuart, 2025-10-03:
I'm not use if it's done to resurrect a 4 year old open issue or start a new one, but I would like to +1 a second AFR table. 

Currently there is a secondary "VE" table only, which is fine if you're using the table as a "Fuel injector duty table", rather than a VE table. If you're are using the VE plus AFR target table, and need to run a second fuel map, It is the AFR table you would want to change rather than the VE, which would/should remain constant for the engine.

For info, my use case is that I will be running a boosted engine with water injection for knock suppression and cylinder cooling. With the VE set correctly for the engine I would like to have a primary AFR target table to maintain an AFR ideal for power (it's apparently ~12.8 for my type of engine) as boost increases and use progressive water injection to do the knock suppression and cooling, rather than running super rich AFRs as many turbo setups do. As a failsafe I would a like to have a second AFR table that would run to these rich AFRs should the water injection fail or run dry. I'm sure there are others with similar use cases that would appreciate this feature.