speeduino/speeduino #599: Feature request for temp delay for VVT

ThresherBuilt, 2021-06-26:
I would like to request a feature be added to delay the VVT from being active until the coolant reaches a certain temp selected by the user. The box can be a “on or off” function so cars who don’t need this feature can leave it off. Once you turn it on there could be a coolant temp box underneath that lights up and the temp can be set by the user. 

Currently my Miata cannot run closed loop VVT consistently because during cold starts the VVT goes crazy due to the high oil viscosity. The Miatas VVT is controlled by oil pressure and oil pressure cold is around 90psi. Allowing the user to define at which coolant temp the VVT goes from OFF to ON would allow us to take full advantage of the use of Closed Loop VVT control.

pazi88, 2021-06-27:
This is actually common issue with pretty much all closed loop vvt systems. Disabling doesn't help much, because the engine still runs like crap, because wrong AFR etc. when VVT is not in use. For time being, you just need to run low enough PID gains that don't cause problems with cold oil. 

OEM solution is to have the closed loop VVT parameters oil temp dependent and this is something I'm working on.

ThresherBuilt, 2021-06-27:
Thanks for responding. I know of a couple ECUs that delay this based on coolant temp and it works well. Like I said this could be a selectable option. Not everyone has oil temp sensors (in fact every Miata owner would have to install an aftermarket one and wire it to the ECU somehow) not sure if that would be the ideal way to go, at least for those running speeduino on Miatas. My VVT car is currently tuned completely without VVT on. I thought the “change fuel with cam angle” setting compensated for the extra fueling needed for the cam advance

pazi88, 2021-06-30:
If there is no oil temp sensor, clt can be used too as substitute. The clt and oil temp follow each other pretty well. “change fuel with cam angle” changes the injection timing so that fuel is injected same time relation to valve opening.

shiznit304, 2021-07-15:
Perhaps a percentage taper kind of like how AE works in the cold

ThresherBuilt, 2021-08-01:
I saw one guy create a code to taper the PID settings based on CLT. That seems a little complex, delaying all VVT advanced until the engine reaches operating temp would be easier and works for megasquirt.