speeduino/speeduino #611: BMP280

emretunali, 2021-07-16:
Hi,

Can we use an advanced barometric sensor like bmp280 instead of mpx?

VitorBoss, 2021-07-16:
The datasheet have code samples for the measurement in float and integer, speeduino only use integers. The SPI isn't possible due some boards use some SPI pins for other stuff, but it can be done for ARM variants of code, for AVR you can do it using I²C.

I don't see a problem with including that option, @noisymime opinion can be different, but you may do the code itself or ask for someone to do it for you if you want to play with it as this is very unlikely most users would use.

emretunali, 2021-07-16:
I'll code and test it myself. @noisymime  can add it to the new version if he sees fit.

noisymime, 2021-07-16:
Happy to look at it provided the code is neat and sane, though I don't believe there is any inherent advantage of a sensor such as this over the MPX or AD234. Given it's onboard, the baro sensor really has no issues with noise or ground offset, but happy to look at including it. 

I would recommend using SPI over I2C as the latter can cause issues with the timers. 

Also worth noting is that this sensor is not automotive rated.

emretunali, 2021-07-16:
bmp280 just wanted to give an example.  Nxp company's FXPS7115D4 automotive digital barometric sensor can be used.  It is newer technology and more affordable than the Mpx series.  Do you want to use SPI or i2c? @noisymime

noisymime, 2021-07-18:
SPI would be preferable. Like I said, no issues with this coming in, but other than widening the range of usable hardware I don't see there being any actual advantage for these sensors. They're all the same capacitive to voltage type sensors, the only difference with these ones is that they include the ADC onboard with a digital interface.

closed 2021-09-21 by emretunali