speeduino/speeduino #613: Idle Up Output delay and PW increase for few seconds

choyr, 2021-07-18:
For Idle Up, can you add a feature to increase the PW to richen up the AFR & increase ignition advance for few seconds, and put a delay to the  idle up output. This is the strategy for Honda OEM ECUs since the ECU controls the AC. When the signal from ACS (AC Switch) activates, ECU increases the pulse width and ignition advance for few seconds, and then delay the output of ACC (AC clutch) to have a smooth AC control.

shiznit304, 2021-07-19:
you can do this with with programmable output. 
[image]

I press ac button, it immediately adds 22% IAC duty cycle, adds 100rpm to idle target rpm. Then after 0.2 s the A/C relay kicks on.
Idle Advance will automatically increase timing to meet my new RPM target quickly. Nothing is done to fueling 
My ac works perfectly like this in my 2000 Miata. The Idle Up RPM Adder is one of my pull requests.

choyr, 2021-07-20:
Nice. Thanks man, so we have to disable the idle up output. I will try that. Thanks man