speeduino/speeduino #628: Feature request: Variable launch rpm

runesmdk, 2021-08-05:
Let the rpm for launch be either locked to a number as it is now, or configurable to lock onto current rpm.

(Can be so activating launch below x rpm is stock behaviour to a tune set rpm, and activating above x rpm makes it lock into current rpm)

This makes for 2 things; quicker launch adjustments for different levels of traction by simply holding the desired rpm and hitting launch button. And also works as rolling launch control for roll racing

Fera31, 2021-10-01:
i have a code with anti lag system with 2nd fuel and 2nd spark for control. and variable launch control. but i don't know how make a git for upload changes.

bluegizmo83, 2023-10-05:
> i have a code with anti lag system with 2nd fuel and 2nd spark for control. and variable launch control. but i don't know how make a git for upload
> 
> [image]

Hey, did you ever get that modified code uploaded on github or anywhere else? I'm definitely interested in seeing those code modifications and trying the variable launch RPM.

runesmdk, 2024-01-29:
> i have a code with anti lag system with 2nd fuel and 2nd spark for control. and variable launch control. but i don't know how make a git for upload
> 
> [image]

Are you still working on/with it?