speeduino/speeduino #632: Possible feature request / or advice

Alexberry10295, 2021-08-14:
Iv purchased my first speeduino to use on my is300 (2JZ VVTI) to which i'm very excited about (so much potential for my self with learning and the platform itself) the UA4C is a capable board, and with the recent VVT improvements im looking forward to learning and tuning.
Currently the 2JZ VVTI 3 tooth cam trigger pattern is not supported. I was wondering if this trigger pattern has been considered?
the 2JZ-GE VVTI shares the same sensors as the GTE VVTI, 36-2 crank and an equal space 3 tooth cam trigger on the intake cam.

Do I grind 2 respective teeth off my kelford cams to suit the firmware or do I wait?
Please and thank you.

pazi88, 2021-08-14:
If you want to have vvt support now, you need to cut off 2 of the cam tooths. The 3 equal spaced tooths are bit weird, because I don't think you can make out sequential sync from that. It's probaply only used for vvt.

mike501, 2021-08-14:
If the crank is 36-2 then a combination of crank and cam could get your
sync with custom code - eg 5 teeth past gap on crank and pulse on cam would
be rotation 0, 30 teeth and cam pulse would be rotation 0, 12 teeth on
crank and pulse is rotation 1.

You'd have to pick one of the combinations to drive vvt in speeduino.

(Made up positions before someone corrects me)

Alexberry10295, 2021-08-19:
> If the crank is 36-2 then a combination of crank and cam could get your sync with custom code - eg 5 teeth past gap on crank and pulse on cam would be rotation 0, 30 teeth and cam pulse would be [...]

thanks for the reply! see custom code is out of my depth, im mechanically minded and can tune my own car so it performs well and reliably. But custom code I dont think is something I can achieve any time soon. but from the knowledge i do have what your are saying should make perfect sense, I havent personally calculated anything with the trigger teeth yet.
Are you savvy or a beginner with custom code?

Alexberry10295, 2021-08-19:
> If you want to have vvt support now, you need to cut off 2 of the cam tooths. The 3 equal spaced tooths are bit weird, because I don't think you can make out sequential sync from that. It's probaply only used for vvt.

its a tempting fix, which I think I would be happy to do on my stock cams. how ever in the very near future ill be running Kelford cams on my new engine :\ so thats concerning. 
Do you know why maybe 2JZ's arent hugely recognised on the speeduino platform?

mike501, 2021-11-06:
A quick hack has been done so it checks for sync on the cam if the crank is within 4 teeth of max teeth. This means it only sees the tooth near the gap on the crank so can positively locate & set Revolution flag.

Now need to figure out how that works with vvt (as Ive never looked at that code)

pazi88, 2021-11-06:
Vvt for missing tooth just checks the crank degree when tooth is detected. So evenly spaced cam tooth trigger is easily done.