speeduino/speeduino #636: VVT rendered inactive in 202108

ThresherBuilt, 2021-08-18:
I was super excited about the new firmware and the updates to VVT. Unfortunately, there’s a bug that prevents the VVT from working AT ALL. 

The new code allows you to designated the cams range of use with Min/max cam angle settings. However, the new code limits you to 0°-100°. The setting states “if the cam angle exceeds this number, it triggers VVT error state and forces DC to 0%. The problem with this is that at 0% duty cycle my intake cam is at 281°. The operative range on a VVT miatas intake cam is approx 280°-315°. The “VVT ANGLE” gauge on the dashboard (and datalog) show that the cam is, in fact, at 281° at 0% DC. 

So, since I can only set the range between 0-100 and my working range is 281-315 the computer thinks that it’s outside the parameters and is therefore putting it in VVT error state, forcing DC to 0%. I can confirm this because my hood duty is set to 37% and on the prior firmware, that where my VVT duty cycle gauge and datalog showed it to be. Now they say 0%. 

So closed loop VVT is completely unusable in this current firmware, worse than the last.

noisymime, 2021-08-18:
What value do you have in the `Cam angle @ 0% duty(deg)` setting? It sounds like you haven't changed it to match the cam (Which would be 281 in your case)

The Min/Max values cover the range of movement, which in your case is only 35° (315 - 280 = 35), but you need to have adjusted the `Cam angle @ 0% duty(deg)` setting to bring your movement into the 0-100 range. 

The VVT angle reading shows the amount that the cam has moved from its base position. So a reading of 0 means that the cam has moved 0 degrees from it's 'rest' position of 281

VitorBoss, 2021-08-18:
I think oyu missed the **%** mark on the min and max values, they are % values because not every valve responds in a 0 to 100% range, but might respond between 37 to 79% range, that setting is to keep the PID in the usable range.

Like Josh said, the option you are after is the `Cam angle @ 0% duty`

pazi88, 2021-08-18:
Also the integer PID loop in speeduino FW craps out if you have non-positive cam angle values (0 not being one). So actual usable cam angle range is 1-99 currently. Use the "Cam angle @ 0% duty(deg)" setting to get into that range as said before.

I did write info on speeduino wiki: https://wiki.speeduino.com/en/configuration/VVT

ThresherBuilt, 2021-08-18:
My cam at 0% is set to 281. The usable cam range needs to be 281-315 like I said. 
[image]

ThresherBuilt, 2021-08-18:
For some reason I can’t post a picture. My VVT settings are all correct. Hold duty set to 37%, cam angle at 0 set to 281. Valve duty (haven’t found the exact range) is 38-79.

pazi88, 2021-08-18:
What is your cam angle reading in open loop mode and in closed loop mode when you use that  "cam angle at 0% duty setting"? Also did you read the wiki about setting up vvt?

ThresherBuilt, 2021-08-18:
It sounds like you guys didn’t actually read my post. Maybe I wrote something incorrect. The VVT angle gauge shows 281!!!! The VVT DUTY CYCLE gauge shows 0%. That is because the ECU has put it into “VVT ERROR MODE” since 281 is outside of the limited 0-100 range. My hold duty is set to 37%. My cam angle at 0 is 281. In case there was any confusion before, all my settings are correct.

ThresherBuilt, 2021-08-18:
Forgive me if I sound harsh. I was sold this ECU as one that would run my VVT in my Miata. I’ve had it for 6 months and it has never run my VVT. I’ve begged and pleaded to have features added and the firmware fixed to run VVT. I was very excited to use my VVT with those update. But it doesn’t work. And others have said the same and had this exact same issue. One guy downgraded back to the old firmware and is patiently waiting for the next release in 6 months to hopefully use his VVT.

ThresherBuilt, 2021-08-18:
My cam angle at 0% is 281°

pazi88, 2021-08-18:
Yes the old VVT implematition was changed because it worked in very few setups. And not even really great in those. So it has been made more universal. Anyways as said you need to use open loop mode to define what is your cam angle at 0% duty. If it is 281, fine. Put that to the setting and when you switch to closed loop, it should show cam angle being 0. Then play around with the "cam angle at 0% duty" setting little bit to get the cam angle to 1-99 range. And you are done. Then it's just up to set the vvt target table.

pazi88, 2021-08-18:
And to be precise. Use cam angle gauge in open loop mode to see what is your cam angle at 0% duty. The put that as the setting. And switch to closed loop.

ThresherBuilt, 2021-08-18:
My cam angle is 281 at idle in open or closed loop with 0% DC

ThresherBuilt, 2021-08-18:
I made this post because everything you’re saying is not working. So obviously there’s a bug. I did all that.

pazi88, 2021-08-18:
If the cam angle reading isn't updated when you adjust "cam angle at 0% duty" setting, it usually means that your cam isn't triggering correctly. Could be something wrong in your setup itself. Does your cam angle reading change when you adjust duty in open loop mode? If the cam angle changes correctly in open loop mode, you need to provide your tune, log and setup details to get that working. In a matter of fact, better to provide those anyways.

ThresherBuilt, 2021-08-18:
I was able to control the cam in the 202103 firmware: this is new to this firmware. When I was on 30 weight break in oil I actually had the VVT working, giving consistent and repeatable results for a full day. The next day it was different. The VVT ANGLE gauge read accurately at that time. Now it says 281 at 0% duty cycle. It never goes into duty cycle because it’s locked to 0 by yo limited 1-99 parameter. This was not a troubleshooting exercise lol. I’m reporting an issue with the firmware. Hence why I posted in the “issues” section. The tune is correct. The physical setup is correct. Something is wrong with the firmware. I’m not the only one with this problem

ThresherBuilt, 2021-08-18:
Y’all’s refusal to acknowledge that possibility is astounding

pazi88, 2021-08-18:
Yes there can be bug. No question about it. But if the bug in question hasn't popped up in release testing phase, but now in your setup for example. You need to provide more details about it so the bug can be reproduced and fixed. "Not working" is not enough information. So tune, exact setup info and logs please.

ThresherBuilt, 2021-08-18:
01 Miata engine with VVT. Factory trigger setup, 4 tooth crank wheel, unmodified cam trigger. Not sure what else you want as far as setup. 350whp turbo, built bottom end. Not sure how you want me to post the tune and datalog

pazi88, 2021-08-18:
Ok, so Miata 99-05 trigger pattern. Not missing tooth. For that only thing added since previous FW release should be cam angle filtering. I personally was too afraid to touch the VVT for that trigger pattern because I don't own a car to test/verify, so I left that as is.

Anyways, I did some digging on older commits and not sure how the closed loop VVT could have ever worked for Miata 99-05 trigger pattern, if it's now showing 280°-315° cam angle. Previously cam angle was only uint8, which has range of 0-255, so the value was probably overflowing in some point. Also digging more of the code, it looks like the "Cam angle @ 0% duty" setting is actually never used for Miata 99-05 trigger, so that explains why the setting doesn't affect. @noisymime Could shed more light into this, because he knows that trigger pattern way better than I. but I think that setting needs to be taken into calculation for Miata 99-05 trigger too or change the fixed 370 adjustment value in code so that the cam angle range is in within 1-99 that works.

ThresherBuilt, 2021-08-18:
Okay. So what you’re telling me is the people who sold me this ECU saying “it will run VVT on a Miata” we’re lying. So I bought into something for 1 reason and that was a deception. Cool. I’m going to just switch to another platform because that’s not cool at all

VitorBoss, 2021-08-19:
If you know how to compile an arduino project we can help you change the single line of the code to make it work, it is really simple, like Pazi sayd, that pattern isn't part of my or his testing, we made mostly recent changes on the VVT PID code.

noisymime, 2021-08-19:
> Okay. So what you’re telling me is the people who sold me this ECU saying “it will run VVT on a Miata” we’re lying. So I bought into something for 1 reason and that was a deception. Cool. I’m going to just switch to another platform because that’s not cool at all. Now it makes sense why I could never get a straight answer from any of those guys and it’s always a bunch of BS.

Your drama doesn't help and won't make it any easier to provide support here. 

The Miata VVT has worked on previous firmwares, I've run this successfully myself and can confirm that it didn't have the issues you describe. It was the first cam angle decoder written though and was done so prior to many (most?) of the changes that have been added since for other patterns. If there are issues with the newer firmware/s then we will work through them and get them resolved

At a quick glance at the Miata cam code, it does indeed look like it was written prior to the addition of some of the newer VVT settings and hasn't been updated. I don't have a BP VVT engine to test on at the moment, but this looks like it should be a relatively straightforward fix. 

If you're willing to try dev code, we can likely get things up and running fairly quickly.

noisymime, 2021-08-19:
> ..., it looks like the "Cam angle @ 0% duty" setting is actually never used for Miata 99-05 trigger, so that explains why the setting doesn't affect. @noisymime Could shed more light into this, [...]

The Miata cam decoder predates that setting and instead used `vvtCLMinAng` for roughly the same thing. Unless I'm missing something, this should be as simple as swapping out `confgiPage10.vvtCLMinAng` for `configPage10.vvtCL0DutyAng` on line https://github.com/noisymime/speeduino/blob/95b8e6ee75f296cde2f2355096d8609b2c39f535/speeduino/decoders.ino#L2300

pazi88, 2021-08-19:
Yes true. That should do it.

ThresherBuilt, 2021-08-20:
OK, so is there a way for me to just get the whole entire Firmware in code form, and then just change that one little line and then upload it into my Speeduino? I will do some research and watch some videos, I assume you use the Arduino app. Just not sure how to physically upload the firmware to the ECU once it’s all put together. I can also use this opportunity to have someone code a coolant temp delay for VVT for me as well

noisymime, 2021-08-20:
If you can hold off a day or so I will test this propose change and get it committed. Once this is done, you'll be able to load the `master` version of the firmware through SpeedyLoader and try this yourself.

ThresherBuilt, 2021-08-20:
Oh okay that would be great. I’m sick on having no torque until 5000rpm. I suspect the VVT will greatly smooth my transition to power.

ThresherBuilt, 2021-08-20:
@pazi88 you mentioned some things about the code for 99-05 Miata trigger pattern that I wanted to find out more about. You said earlier that you’re not sure how it could ever work because something about the range was 0-280 and Miatas range is 280-320. Once the “cam angle at 0%” setting is fixed is this going to run my VVT or is there another issue as well? You said there were no other changes to the Miata VVT besides the cam filtering so am I to assume it will function the same as it did on the 202103 firmware?

My VVT Is not usable on the 202103 firmware. It’s unpredictable and inconsistent and I have it turned off. I was banking on this update to solve the issues that made closed loop VVT so bad, because I read something on here about some fixes that were created a month or two ago which made this more stable. It didn’t specify Miata, but I assumed since the VVT setting in TunerStudio specifically says “VVT is only for Miata trigger patterns” that the Miata is the only car able to run VVT currently (202103 FW)

Thanks for all your help

pazi88, 2021-08-20:
The closed loop VVT fixes affect all the trigger patterns. So when the “cam angle at 0%” setting works and you get the cam angle to usable range, the VVT control should work much better. Indeed it was very inconsistent previously. I could not drive my car with closed loop VVT at all. But with these new fixes I have daily driven my car for months without any closed loop VVT problems.

noisymime, 2021-08-20:
OK, with that last commit you should be able to try this. If you open SpeedyLoader and scroll to the bottom of the firmware list, you should see the `master` entry. If you choose this and load it then you should be able to give this a try. 

(You'll need to load the dev ini file as well, which SpeedyLoader will put in your downloads folder)

ThresherBuilt, 2021-08-20:
Ok great, thanks for fixing that so quick and making it happen. I’m grateful as I’m sure many others will be.

ThresherBuilt, 2021-08-21:
When you said speedyloader would put the ini in my downloads folder, you mean when I connect the laptop to the ECU and load the master firmware onto the ECU it will also trigger a download onto the laptop itself?

ThresherBuilt, 2021-08-21:
Alright I have successfully loaded the new firmware onto my ECU and updated the ini in TunerStudio so I will report back tomorrow with my experience. Thanks again

ThresherBuilt, 2021-08-21:
IT WORKS! The VVT is giving me consistent and repeatable results. I’ve only driven the car once with these settings, previously it was inconsistent across runs so Well see but right now it’s on point. Good job

DirtyPeanut1, 2021-08-21:
I think some serious thank you’s and apologies to all parties involved
would be in order.

debrucer, 2021-08-21:
You’ve got that right. It was painful to watch and all respect to those who
provided the solution.

ThresherBuilt, 2021-08-21:
I’ve been getting brushed off and ignored by people for 6 months about this VVT thing. I started off trying to help be part of the solution, and after soooooo many times I started to get irritated. I said thanks like 3x already, and I’ll say it again, thanks. However, had I not raised such a stink about it, all the Miata owners probably would have waited another 5-6 months for the next update before anything happened. So thanks again for your help, sorry I had to get so assertive about it. The results speak for themselves. Good team work

noisymime, 2021-08-21:
Glad to hear things are working well now. 

When these sorts of things happen, absolutely discuss it on the forums, Slack, Facebook whatever, but if you feel there's an underlying issue that's not being addressed, your best bet is always to raise an issue like this. The other devs and myself don't always see all the conversations going on, but at the very least I will always look at Github issues. 👍

closed 2021-08-21 by noisymime