speeduino/speeduino #638: feature request: PWM IDLE Duty Cycle Multiplier

choyr, 2021-08-18:
In my Honda B20 with a 500Hz IACV, a value of 1% has a change of about 150 rpm which is too coarse. Current setting is only 88-89%, with additional 4% from idle up. If possible to add a multiplier (ex. 100% is no change, and 10% is 10% of the final Duty Cycle) to have a better resolution and control.
Thanks
[image]

shiznit304, 2021-08-21:
did you add a flyback diode across your IAC

shiznit304, 2021-08-22:
I had the same issue with my NB1 Miata. You just need to solder in a diode across the 12v and ground wires in a certain way. That brought my pwm from the 86-89 range down to 20 - 60 range. It also reduced the audible buzzing noise

choyr, 2021-08-22:
Alright. Let me test it, because i did not put a flyback diode.

closed 2024-03-15 by noisymime