speeduino/speeduino #642: TPS DOT as condition for programmable output not working

shiznit304, 2021-08-22:
[image]
TPS DOT as a condition appears to do nothing. I want to use TPS DOT to trigger the opening of my supercharger bypass valve.

shiznit304, 2021-09-11:
I believe Issue #654 is related to this. Perhaps I need to select the item above TPS dot to get it to work. @morrisman1 @afroboltski

shiznit304, 2021-09-20:
I was able to get TPSDOT to trigger my programmable output but it's not reliable. Maybe the condition check refresh rate is too slow to catch tpsdot reliably?

VitorBoss, 2021-09-20:
The update rate is 10x/second for all checks. TPSdot is on the list but it can update faster that 15Hz, I'm using it at 30Hz rate

closed 2021-10-13 by shiznit304