speeduino/speeduino #664: Feature request: Felicia/Favorit trigger pattern

kovama1, 2021-09-21:
Hi guys,

could someone create trigger pattern for Škoda Felicia/Favorit? Its taken from flywheel and its quite hard to mount alternative trigger on crank pulley. It already have hall sensor from factory so it would be neat to just use it :)

Thanks a lot! ¨

![trigger_Felicia_Favorit](https://user-images.githubusercontent.com/29120858/134138579-4a5dedc7-551f-4238-b815-1b311f35aff4.png
[image]
[image]
)

digmorepaka, 2021-09-21:
The Felicia and Favorit used a distributor for ignition, this was because this crank trigger has extremely low resolution and is only really good for timing singlepoint injection(which is what it was originally for). You should mount a regular missing tooth trigger into the distributor and run a Golf MK4 1.6 benzin waste spark module(or similar) or coil on plug instead of the oem system that would be gutted.

kovama1, 2021-09-21:
> The Felicia and Favorit used a distributor for ignition, this was because this crank trigger has extremely low resolution and is only really good for timing singlepoint injection(which is what it [...]

No, it is for the revised version with Siemens electronics, multipoint injection and COP-like setup. Low resolution is not an issue for me and I can live with that (Like with 4G63 trigger on miata and many others...). It's just a few lines of code which could save someone spending quite a lot of money for buying sensors and developing a custom pulley to adapt a custom trigger wheel. 

[image]

digmorepaka, 2021-09-21:
I believe the 4G63 and MX-5 triggers work because while still low resolution they provide the position at far more accuracy because of multiple edges. You have mentioned the Favorit and Felicia together, so i assumed it was the Monomotronic SPFI. Are you sure the trigger wheel stayed identical between MPI and monomotrinic?

kovama1, 2021-09-21:
> I believe the 4G63 and MX-5 triggers work because while still low resolution they provide the position at far more accuracy because of multiple edges. You have mentioned the Favorit and Felicia [...]

Yup you are correct. For favorit (bosh monotronic) we dont have that one addtional teeth. 

So sorry i only need trigger pattern for Felicia with addtional teeth ;)

For reference there is Favorit flywheel: 
[image]
[image]

mike501, 2021-11-06:
Do you know if it uses a hall effect or vr sensor? If it's hall effect you can double the edges by looking at the rising and falling edge.

kovama1, 2021-11-06:
> Do you know if it uses a hall effect or vr sensor? If it's hall effect you can double the edges by looking at the rising and falling edge.

Yup it is hall effect sensor ;)

mike501, 2021-11-18:
I'm absolutely not committing to writing this. But if some code turned up to support the engine, when would you be realistically ready to test it? Would it be in a day, week, month, 6 months etc

kovama1, 2021-11-29:
> I'm absolutely not committing to writing this. But if some code turned up to support the engine, when would you be realistically ready to test it? Would it be in a day, week, month, 6 months etc

Probbably week. I have acces to this car only at weekends ;)

3pdp, 2023-07-07:
Hi, im in process of building a turbo felicia with speeduino, I've noticed rusefi has support for felicia mpi trigger wheel and wondered if speeduino will ever support it or should I look for an alternative solution?
Thanks

Bobika2002, 2023-08-22:
Any news about this? Im thinking about same thing.

kovama1, 2023-08-22:
> Any news about this? Im thinking about same thing.

Well seems like no one from devs is catching up on this :/ 

But I looked on Daihatsu +1 trigger and it seems like it would be easy to modify it to work like felicia one:
[https://wiki.speeduino.com/en/decoders/Daihatsu_+1]

Mabye just changing this to different tooth count and angles should do the trick:
```
 if(configPage2.nCylinders == 3)
  {
    toothAngles[0] = 0; //tooth #1
    toothAngles[1] = 30; //tooth #2 (Extra tooth)
    toothAngles[2] = 240; //tooth #3
    toothAngles[3] = 480; //tooth #4
  }
  else
  {
    //Should be 4 cylinders here
    toothAngles[0] = 0; //tooth #1
    toothAngles[1] = 30; //tooth #2 (Extra tooth)
    toothAngles[2] = 180; //tooth #3
    toothAngles[3] = 360; //tooth #4
    toothAngles[4] = 540; //tooth #5
  }
}
```

mike501, 2023-08-22:
I only write decoders during winter in the UK. I'm still supporting decoders I wrote last winter + one I foolishly took on a couple of months ago. If its still required in a few months I may have a look.

If you want to use the above decoder with modified angles, you'd want to set it to have 3 teeth / toothangles, the two proper teeth + the third +1 tooth (i'm assuming you're going to pick a specific rising or falling edge, if you want to use "change" it would be 5 teeth but remember this only works for hall sensors.)

Tomika012345, 2024-08-08:
Hi, any news in this topic? Is the Daihatsu patter working for this? 

Thanks