speeduino/speeduino #673: Aux Inputs Not Working

github-account-0, 2021-10-03:
I am unable to get aux inputs working using the latest ini file and compiling the latest firmware (1d3a30ab4c08b146dd92ea21c55c9f138f808be8).  I have also applied fix #655 and no difference.  I first noticed this issue when trying to test PR #586 

There is one sequence I can run through to get the inputs to behave as expected which I will post below but the problem is as soon as I power cycle the input is unresponsive again.

> 
> Then there is a series of events i can do to make it work briefly
> 
> I go to Secondary Serial IO Interface
> Enable Second Serial
> 
> Then go to local auxillary input configuration
> change aux input 0 to external source
> click burn and power cycle
> 
> then i go back to local auxillary input configuration
> change aux input 0 to analog local a8
> click burn but do not power cycle
> 
> during this period the input works as expected but as soon as i power cycle it is back to not working

Attaching images of my settings and ini file/relevant inf0

[image]
[mainController.zip](https://github.com/noisymime/speeduino/files/7274149/mainController.zip)

shiznit304, 2021-10-03:
This is the fork I currently run in my 2000 Miata, feel free to give it a try. My aux inputs work fine. https://github.com/shiznit304/speeduino/commits/mycar5 
It's the main branch from August 20, 2020 plus a few PRs such as #608 #549 #586 #607 #649 and one PR i haven't published yet for idle up with closed loop

Afroboltski, 2021-10-03:
Ah ha. I see the problem. https://github.com/noisymime/speeduino/issues/654 was closed and "fixed" but whoever merged it only applied half my fix - Aux in1-16 should be Aux in 0-15 in the fullStatus defs at the top.

github-account-0, 2021-10-03:
> Ah ha. I see the problem. #654 was closed and "fixed" but whoever merged it only applied half my fix - Aux in1-16 should be Aux in 0-15 in the fullStatus defs at the top.

I tried simply renaming them and it didnt work, is my PR #675 wrong then and there is more to it?

github-account-0, 2021-10-03:
I was able to workaround this bug by modifying the following line in globals.h
https://github.com/noisymime/speeduino/blob/93de5eecda3bc288394da73b998331080694e877/speeduino/globals.h#L607

to read

`#define pinIsUsed(pin)      ( pinIsInjector((pin)) || pinIsIgnition((pin)) || pinIsSensor((pin)) ) //|| pinIsOutput((pin)) || pinIsReserved((pin)) )`

github-account-0, 2021-10-03:
Im confused because A8 is free in the init I even commented the spare1 out and it had no effect
```

    case 41:
    #ifndef SMALL_FLASH_MODE //No support for bluepill here anyway
      //Pin mappings as per the UA4C shield
      pinInjector1 = 8; //Output pin injector 1 is on
      pinInjector2 = 7; //Output pin injector 2 is on
      pinInjector3 = 6; //Output pin injector 3 is on
      pinInjector4 = 5; //Output pin injector 4 is on
      pinInjector5 = 45; //Output pin injector 5 is on PLACEHOLDER value for now
      pinCoil1 = 35; //Pin for coil 1
      pinCoil2 = 36; //Pin for coil 2
      pinCoil3 = 33; //Pin for coil 3
      pinCoil4 = 34; //Pin for coil 4
      pinCoil5 = 44; //Pin for coil 5 PLACEHOLDER value for now
      pinTrigger = 19; //The CAS pin
      pinTrigger2 = 18; //The Cam Sensor pin
      pinTrigger3 = 3; //The Cam sensor 2 pin
      pinFlex = 20; // Flex sensor
      pinTPS = A3; //TPS input pin
      pinMAP = A0; //MAP sensor pin
      pinBaro = A7; //Baro sensor pin
      pinIAT = A5; //IAT sensor pin
      pinCLT = A4; //CLS sensor pin
      pinO2 = A1; //O2 Sensor pin
      pinO2_2 = A9; //O2 sensor pin (second sensor)
      pinBat = A2; //Battery reference voltage pin
      pinSpareTemp1 = A8; //spare Analog input 1
      pinLaunch = 37; //Can be overwritten below
      pinDisplayReset = 48; // OLED reset pin PLACEHOLDER value for now
      pinTachOut = 22; //Tacho output pin
      pinIdle1 = 9; //Single wire idle control
      pinIdle2 = 10; //2 wire idle control
      pinFuelPump = 23; //Fuel pump output
      pinVVT_1 = 11; //Default VVT output
      pinVVT_2 = 48; //Default VVT2 output
      pinStepperDir = 32; //Direction pin  for DRV8825 driver
      pinStepperStep = 31; //Step pin for DRV8825 driver
      pinStepperEnable = 30; //Enable pin for DRV8825 driver
      pinBoost = 12; //Boost control
      pinSpareLOut1 = 26; //low current output spare1
      pinSpareLOut2 = 27; //low current output spare2
      pinSpareLOut3 = 28; //low current output spare3
      pinSpareLOut4 = 29; //low current output spare4
      pinFan = 24; //Pin for the fan output
      pinResetControl = 46; //Reset control output PLACEHOLDER value for now
    #endif
      break;
```

It is one of these outputs but everything shows board defaults listed above

`#define pinIsOutput(pin)    ( ((pin) == pinFuelPump) || ((pin) == pinFan) || ((pin) == pinVVT_1) || ((pin) == pinVVT_2) || ((pin) == pinBoost) || ((pin) == pinIdle1) || ((pin) == pinIdle2) || ((pin) == pinTachOut) )
`

I cant find A8 being used when looking through menus or at io summary  besides my idle up setting.

github-account-0, 2021-10-03:
Is idle up set as pinIdle1 or pinIdle2?

Is that why it doesnt work?

shiznit304, 2021-10-03:
my guess is that it uses pinIdle 1 in a 2 wire idle controller, and both
pin idles in a 3 wire controller?

Afroboltski, 2021-10-04:
I think half the problem here is that the pin number for various functions is still written to the EEPROM from TS, even if the function is disabled. For example you set idle valve to pin X, then disable idle control, pin X is still being written to the EEPROM, so the firmware still believes it's used?

You'll see in my A/C compressor control implementation (https://github.com/noisymime/speeduino/pull/665), I interlocked the pin assignment with the A/C enable status, but no other feature seems to do this

shiznit304, 2021-10-04:
have you looked into #549 ? I didn't need it to make it work but I did need it to use idlePin2 for other functions other than for idle control. In fact, i use idlePin2 to play euroBeat when TPS > 50% 😂

github-account-0, 2021-10-04:
I tried out #549 it didnt work for me in this case

github-account-0, 2021-10-04:
If anyone else has this issue and needs a WORKAROUND(perhaps not a proper solution) they can try my fork and ini file that has #655, #676, and reverts part of #398

https://github.com/github-account-0/speeduino

choyr, 2022-01-14:
tried it today with UA4C, it is not working for Aux input gauges, but if I use it for fuel/oil pressure, it works

Afroboltski, 2022-01-16:
Yup it's still wrong in the INI file in the current release, and the off-by-one error still exists on line 100 for sensors.ino

Corey-Harding, 2022-01-16:
Check PR #745

Corey-Harding, 2022-01-17:
> tried it today with UA4C, it is not working for Aux input gauges, but if I use it for fuel/oil pressure, it works

What did you try?

choyr, 2022-01-17:
I used A8 to A13 and tried aux input gauges and they are not responding. However, if i use for example A8 as fuel pressure, it works same as A10 as oil pressure. Maybe i did something wrong, i will check again later

Corey-Harding, 2022-01-17:
> I used A8 to A13 and tried aux input gauges and they are not responding. However, if i use for example A8 as fuel pressure, it works same as A10 as oil pressure. Maybe i did something wrong, i will check again later

What fork are you running or what pr are you testing?(was what i meant)

choyr, 2022-01-17:
Ahh. The default firmware 202202. Not yet tested yours. Will try it later

Corey-Harding, 2022-01-17:
Try this first
https://github.com/Corey-Harding/speeduino/tree/ProgOutputPR

Then if it doesnt work try my master
https://github.com/Corey-Harding/speeduino

The only difference is on the master I reverted #398 via 
https://github.com/Corey-Harding/speeduino/commit/e11864160506a487f099fc83d86c1fd4795423fd

choyr, 2022-01-19:
Hi Corey, just tested your ProgOutputPR and everything works fine except A8. for UA4C board, A8 is pinSpareTemp1 and A9 is pinO2_2. AFR2 is available in TS, but SpareTemp1 is not. I tried to use A8 in aux input and it does not work, howevery A10 to A13 works.

shiznit304, 2022-01-19:
I’m using a8 and a9 on my ua4c just fine from local auxiliary inputs menu.
I know in the past they didn’t work but they seem fine. I use them to
monitor fuel tank level and fuel tank pressure.

Corey-Harding, 2022-01-19:
If you scroll up in this issue I couldnt get A8 to work without reverting 398 I even commented out pinSpareTemp1

But maybe try changing pinSpareTemp1 to A10 and see if A8 works

choyr, 2022-01-19:
so I changed pinSparetemp1 to A15 to see if A8 works, and no. A8 still does not work

Corey-Harding, 2022-01-19:
This is what it took for me to get A8 to work on UA4C
https://github.com/Corey-Harding/speeduino/commit/e11864160506a487f099fc83d86c1fd4795423fd

choyr, 2022-01-19:
it did not work for me. i will check it again next time

Corey-Harding, 2022-01-23:
Its been cold so i havent been using my ac but this bug is back

A8 turns on my ac for ~5 seconds then its off for ~10 seconds then repeat on ~5 seconds and off ~10 seconds

My reverting 398 doesnt fix it

Afroboltski, 2022-01-23:
> Its been cold so i havent been using my ac but this bug is back albeit in a different form
> 
> Now A8 turns on my ac for ~5 seconds then its off for ~10 seconds then repeat on ~5 seconds and off ~10 seconds.
> 
> My reverting 398 doesnt fix it and turning idle up off it has the same results

Can you post your tune file? And also what FW and INI file you're using?

I was still convinced in the back of my mind that the problem was the fact that you can set a feature to a particular pin, and then disable the feature, and the box goes grayed out in TS but the tune file still programs the pin assignment into the ECU, and most features look like they don't really interlock pin assignments with whether or not the feature is enabled? So A8 could have been assigned to something and you wouldn't know?

However this latest version of the bug says otherwise.

Corey-Harding, 2022-01-24:
I am running this branch
https://github.com/Corey-Harding/speeduino

Its the same as official master plus pr #745 added and pr #398 reverted
https://github.com/Corey-Harding/speeduino/commit/e11864160506a487f099fc83d86c1fd4795423fd

Will post my tune soon

Corey-Harding, 2022-01-24:
Here is the current tune

[2022-01-23_19.29.17.zip](https://github.com/noisymime/speeduino/files/7921821/2022-01-23_19.29.17.zip)

Corey-Harding, 2022-01-24:
If there was an IdleUpActive True/False status under programmable output options as my trigger instead of me having to use Aux 0 Analog Input A8 I feel like it would solve my issue.

shiznit304, 2022-01-24:
Do what I did, wire your ac input into an op amp, it’ll turn your signal
into an on/off input

Corey-Harding, 2022-01-24:
Im just concerned if I resolder the wire to a different pin or do as ypu say that I still have to use that pin for idle up so I think adding idleUpActive as a trigger instead of reading the 0-1023 value of A8 is a good option. Im not sure how to do that yet though.

shiznit304, 2022-01-24:
I used to use analog input as an AC trigger but unfortunately The voltage
level from the AC switch does not go low enough to trigger a digital low
signal. It wouldn’t go below the threshold, so instead I added an app amp
to turn the 12 V to 3 volt signal into a 0/5 V signal for digital input

Afroboltski, 2022-01-24:
Just wait until my AC control is merged ;)

[https://github.com/noisymime/speeduino/pull/665](url)

shiznit304, 2022-01-24:
You don’t have idle rpm target adjustment in there. I recommend adding that
feature

Corey-Harding, 2022-01-24:
Your PR cant automatically merge or I would test it.  Might check into it still though if I have time to see what is conflicting.

Corey-Harding, 2022-01-24:
Also @Afroboltski i was told not to do +1 on auxinpina here
https://github.com/noisymime/speeduino/pull/676

Afroboltski, 2022-01-24:
the +1 is ONLY for the digital pins, the analogs were OK I believe, from memory

Corey-Harding, 2022-01-24:
You have it in your pr to do +1 on analog as well.

Also you dont have an output selection for condensor fan. I have two outputs i need. One for ac relay and one for condensor fan.

Afroboltski, 2022-01-24:
> Your #665 PR cant automatically merge or I would test it. Might check into it still though if I have time to see what is conflicting.

Yeah it's a bit stale, I'm planning on updating it (just re-writing it will be easier actually) well before the next FW release

> You don’t have idle rpm target adjustment in there. I recommend adding that feature

OK, consider it done!

Corey-Harding, 2022-01-24:
See comment above i feel like we posted at the same time and you may miss it

Afroboltski, 2022-01-24:
> You have it in your pr to do +1 on analog as well.
> 
> Also you dont have an output selection for condensor fan. I have two outputs i need. One for ac relay and one for condensor fan.

Ahh. I see the confusion.

Looking at #676, I agreed with pazi88, who pointed out that adding the +1 to the auxinpinaline was incorrect.

And in #655 I never mentioned auxinpina.

Looks like I chucked it into the Air Conditioning PR (#665) by accident! Sorry about that.

Corey-Harding, 2022-01-24:
Also will you add an additional ac condensor fan output field to your pr for those like me where it isnt the same as the relay

Afroboltski, 2022-01-24:
> Also will you add an additional ac condensor fan output field to your pr for those like me where it isnt the same as the relay

Yes, in fact I already did! As I mentioned in the 13th of October comment 😎

Corey-Harding, 2022-01-24:
Thats awesome news!  Forgive me if I had read it before I had forgotten.

Corey-Harding, 2022-03-16:
I think my aux inputs are working(I only have one on A8) and its my combination of settings that won't work.  I was able to unrevert the pinisused related commit and switch my idle up output to a digital pin and setup same digital pin to an input and now the current firmware works the same as a few weeks ago and ac cycles on and off a few seconds at a time.

I am unsure if the minimum time setting in progout is working though, at least not with my settings as I tried to use that to compensate and it didnt work.

I am also starting to wonder if idle initiliase is triggering every few seconds and reinitialising my pins or another initialise event is occuring.  It seems idlue up steps does not work in ol+cl so not sure what else is missing as well.

Afroboltski, 2022-07-09:
Dredging up the past here,

Does this fix this issue? #887

noisymime, 2024-02-05:
Closing as this appears to have been resolved quite some time ago

closed 2024-02-05 by noisymime