speeduino/speeduino #697: Feature request: Accel enrichment - Factor in "start load"

theKraid, 2021-11-03:
The current accel enrichment implementation in Speeduino is adding fuel by a TPSdot-based **factor of current VE** (and then of course tapered by RPM)
However it is missing a "start load" factor, which is present in other ECUs in different ways. This factor is important because for example when going from 50% to 100% throttle you need considerably less additional fuel than going from 0% to 50%.
The most comprehensive visual example of this would be this table from a commercial ECU which is essentially the same than the one in Speeduino but with the addition of the "start load" axis:

[image]

Another example would be MS2, where the added fuel is a configurable pulsewidth in ms, which from my understanding in effect is similar in that it makes a bigger difference in low load than high load.

We had this discussion with Josh in the FB group and I thought moving it here would be a better way to keep track of it. One idea of an easy implementation he mentioned without having to resort to expanding the table into 3D would be to simply add another taper, functionally similar to the existing rpm taper, but considering start load instead.
Things to consider:
* By "Start Load" we probably want to reference not the current load, but the one in the cycle before the AE set in. For example if you use tps-based load like I do and change from 0% to 100%, the moment the AE sets in it would essentially not add anything due to the taper. Luckily according to Josh that already exists: "It already tracks to the AW start load because it can actually change during the TPS push. IE you press the throttle but then press it harder part way through its travel, increasing the AE amount needed"
* The implementation should be tested by both TPS-based and MAP-based setups

I could never set up the AE in my cars with ITBs up "just right" and I believe this is the core issue. I think this could potentially be a major improvement. If anyone would be willing to implement this I would be very grateful.
Cheers and keep the great work up!

VitorBoss, 2021-11-03:
> Another example would be MS2, where the added fuel is a configurable pulsewidth in ms, which from my understanding in effect is similar in that it makes a bigger difference in low load than high load.
> 

This is already added to speeduino, use `PW Adder` instead of `PW Multiplier`, this way will apply your req_fuel % to the final PW, I'm using it for more than a year

pazi88, 2021-11-03:
> This is already added to speeduino, use `PW Adder` instead of `PW Multiplier`, this way will apply your req_fuel % to the final PW, I'm using it for more than a year

I have been also using the PW Adder mode. Works much better than multiplier.

theKraid, 2021-11-03:
> I have been also using the PW Adder mode. Works much better than multiplier.

Agreed, PW Multiplier never worked well for me either.
That said, as far as I can tell, even the `PW Adder` function is not a fixed value, but a factor of VE / req_fuel:
https://github.com/noisymime/speeduino/blob/b0acb927429b864b9b19b2c0025ac3f3e3766585/speeduino/speeduino.ino#L1330-L1332

I might be wrong. But regardless, I still think this feature would make a great difference

VitorBoss, 2021-11-03:
On that line `intermediate` is the final PW, so it uses the AE amount % of `req_fuel` to be added to the PW. AEamount at 100% mean no change, that is why it subtracts 100 from it, so that line does `(5000 * (105 - 100)) / 100 -> (5000 * 5) / 100 -> 25000 / 100 = 250` so 5% of 5000 is 250, what is expected.

theKraid, 2021-11-03:
Okay my bad, I understand. However since we know that req_fuel is factored in, the AE scales with VE instead of fixed pulsewidth lengths, so we know that there is no indirect start-load taper as theorized in my original port. Which would mean that the idea of adding a Start-Load taper is still valid, right?

theKraid, 2025-11-24:
This is still relevant - I firmly believe the quality of AE would be massively improved by factoring in Start-load.
As a suggestion: If flash space is a concern, this feature does not necessarily have to be implemented by adding another axis to the AE table. A linear factor would be a close enough approximation. To keep compatibility to existing tunes that tuned around this issue, an on/off toggle could be implemented in the AE dialogue