speeduino/speeduino #711: idle.ino

AmesisProject, 2021-11-17:
Hello, I am not sure because I am new to code, but I do not understand why there is a second "case 2" in the file idle.ino line 150.
I think it should be the 4. If it's the 4 I can pullrequest the correction.
Thank you
[image]

github-account-0, 2021-11-17:
You are right but that is just a comment (noted by the "//") with a wrong or outdated number.  Comments are not compiled into the program and does not affect the program in any way.  They are only there as a reference to help programmers understand what a specific piece of code does.

AmesisProject, 2021-11-17:
Yes I know it's a comment. But I wanted to be sure I understood. It wasn't a criticism at all. It is that I explore the code and I try to put together the info.

noisymime, 2022-02-04:
Closing as this is not an issue, just an outdated comment

closed 2022-02-04 by noisymime