speeduino/speeduino #716: Feature Request: Radiator fan indicator

github-account-0, 2021-11-18:
I am working on a pi dashboard using tuner studio and noticed there is no indicator available to see when your radiator fan is running.  It would be nice to have that feature.

I am also curious about adding buttons on the tuner stusio dash to turn on your radiator fan and condenser fan.

npmstart-pray, 2021-11-19:
Since it's possible to have speeduino be a pwm controller for electric cooling fans, might I suggest that along with coolant temp displayed, we can see the duty cycle of the cooling fans? It might make config of Pwm controller a tad more accurate and faster.

pazi88, 2021-11-19:
> Since it's possible to have speeduino be a pwm controller for electric cooling fans, might I suggest that along with coolant temp displayed, we can see the duty cycle of the cooling fans? It might make config of Pwm controller a tad more accurate and faster.

PWM fan control and duty display here: https://github.com/noisymime/speeduino/pull/637

github-account-0, 2021-11-19:
> PWM fan control and duty display here: #637

I don't have a strong need for a pwm controlled fan. I just want to show if my rad fan is on or off. I might tackle this in a couple months when I have more free time.

npmstart-pray, 2021-11-19:
I’ve always felt my fan comes on too late and has to run too strongly because of it. Being able to adjust on/off and duty cycle would be awesome for me. Glad someone has requested it before me 

Sent from nowhere near my ready room with my new tricorder

github-account-0, 2021-11-20:
In testing the fan indicator on the dash shows on and off as expected alongside the radiator fan switching on and off.

npmstart-pray, 2021-11-20:
Nice!  Now an ammeter reading alternator current output reading would be sweet to verify charging system load. 

Sent from nowhere near my ready room with my new tricorder

mike501, 2021-11-20:
We already have a gauge that shows car system voltage. Doesn't that cover
the alternator? If it's not 14.5 (ISH) volts it's not working properly. If
you wanted a light you could do programmable output to turn on a pin at
lower than X volts.

npmstart-pray, 2021-11-20:
Not voltage, amperage. The current passing through the electrical system. You know the alternator can deliver 125A or whatever at “12vdc”, but that’s max output. If you put an ammeter on the system you can tell how hard it’s working…like when your electric fans are drawing 30A on high 

Sent from nowhere near my ready room with my new tricorder

github-account-0, 2021-11-25:
I merged #719 into my branch https://github.com/github-account-0/speeduino/tree/Radiator_Cooling_Fan_Indicator and I now have a working radiator cooling fan indicator on my tuner studio dash.

commit b419f15a (2021-11-25): Add Radiator Cooling Fan indicator (#719). Fixes #716

* Update auxiliaries.ino

Add Radiator Fan Status Indicator Changes

* Update globals.h

Add Radiator Fan Status Indicator Changes

* Update cancomms.ino

Add Radiator Fan Status Indicator Changes

* Update speeduino.ini

Add Radiator Fan Status Indicator Changes

* Update speeduino.ini

Simplify Naming

* Update auxiliaries.ino

Simplify Naming

* Update cancomms.ino

Simplify Naming

* Update globals.h

Simplify Naming

* Update speeduino.ini

Enable Datalogging of Radiator Cooling Fan Status

* Update auxiliaries.ino

Remove old instanc [... cut]
closed 2021-11-25 by noisymime